xoseperez/espurna #537: Sonoff Dual: additional buttons (or recommendation for other board)

skyynet, 2018-02-07:
I'm quite happy with the Sonoff Dual but would like to read some feedback from the relays actions. Thus I'm looking for a way to add at least two buttons which should be readable from ESPurna.

Is there a way to **add two buttons to the Sonoff Dual** board or is there a **board available with at least two relays and two separate buttons** (not the ones which switch the relays)?

An additional sound recognition module would be the final kick (I know, I'm a dreamer).

I use the module to open/close a "garage" blind for my robomow by schedule.
Now I would like to close the blind when the mower entered the garage (contact at charging bay) and open it again when the mower announces its departure by multiple beeps (this could be done with a separate "clapper" module or - more sophisticated - sound recognition chip.

(any way I can tag this as a question and not an issue while creating a new issue? Dunno how)

xoseperez, 2018-02-08:
The dual has headers on the board to add two additional buttons. Actually, it's the only Sonoff with this feature (AFAIK).

As for the clapper... I'm afraid you will have to locate a spare GPIO and solder a thin wire to it (maybe directly on the chip)...

skyynet, 2018-02-11:
Thanks! So I can connect GND with Button0 or Button1 and read these values from ESPurna?
Do I have to configure anything else than just define BUTTON1_PIN 0 and BUTTON2_PIN 1 to hardware.h Sonoff_Dual section? As there is already a comment #define BUTTON3_RELAY 1 I assume that the button1 header will switch the relay? I think, I will just have to test this...

skyynet, 2018-02-11:
Just tried to configure
    define BUTTON1_PIN         0       // Button 0 on header
    define BUTTON2_PIN         1       // Button 1 on header
    define BUTTON1_MODE        BUTTON_PUSHBUTTON
    define BUTTON2_MODE        BUTTON_PUSHBUTTON

but only got error

In file included from /config/hardware.h:371:5: error: 'define' does not name a type
     define BUTTON1_PIN         0       // Button 0 on header
     ^
In file included from config/general.h:209:5: error: 'custom_reset_hardware' was not declared in this scope
     custom_reset_hardware, custom_reset_web, custom_reset_terminal,

I just want to read the button presses and not switch any relays.

xoseperez, 2018-02-12:
I recommend you to use the definition of the Sonoff Dual in the hardware.h file and maybe change the BUTTON#_MODE for buttons 0 and 1 to BUTTON_PUSHBUTTON. The error you are getting is because #define has a leading hash (#).

skyynet, 2018-02-13:
Argh! Now I see my mistake. As a PHP programmer the # is a comment character and I thought I had to remove it. Now I was able to compile the code and upload it into my Sonoff Dual r1.

I thought that I could see the values of the pushbuttons in the GUI but can't. Also querying via api just tells me the relay status. What do I have to change to see the button status?

skyynet, 2018-02-22:
Since I know now that I can use the two headers for additional buttons, I'd still like to know how I can make use of the pressed buttons, e.g. notice them. I can't find any info in the GUI nor can I query them via API. Help appreciated. I test this currently with a Sonoff Dual R1 but also received an R2 now.

xoseperez, 2018-02-26:
With the original Dual you cannot get more events than button released since the 3 buttons are driven by the secondary microcontroller. With the R2 you can get all events via MQTT in the `<root topic>/button/#` topic (replace `<root topic>` with your device root topic and `#` with 0, 1 or 2 for each button).

This is documented (along with all possible values) in the wiki: https://github.com/xoseperez/espurna/wiki/Buttons-and-switches.

skyynet, 2018-03-03:
Thanks. I thought it might be possible to check the current state via REST API. My initial question is answered so I close this issue. It would be great to be able to check the stats via REST but that's apparently an improvement request and not an issue.

closed 2018-03-03 by skyynet