xoseperez/espurna #538: Frequent crash and reboot

Tito1337, 2018-02-07:
Hi Xose,

I have a big stability issue with Espurna. I don't know how to trigger the crash. Sometimes it is stable for more than 8 hours and sometimes it reboots every 3 minutes. I have 4 Espruna devices running and they often reboot within seconds of each other.

Here is the stacktrace with pre-compiled firmware  **espurna-1.12.3-itead-sonoff-touch.bin**

```
Exception (0):
epc1=0xb1c04012 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd00 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffea0:  db290000 f2301402 1cd83ffe b1c04010  
3ffffeb0:  00003ffe 00000000 b2100000 00003ffe  
3ffffec0:  db290000 21161402 01004010 ffff0000  
3ffffed0:  b1c07fff b1c03ffe 00013ffe 00010000  
3ffffee0:  00000000 2f390000 ea904010 003c3ffe  
3ffffef0:  db290000 00001402 050c0000 c2784000  
3fffff00:  1eb03fff c2004010 00223fff b19c0000  
3fffff10:  07933ffe 00004010 ea600000 c31100fe  
3fffff20:  00304024 001a0000 ffff0000 c2f4ffff  
3fffff30:  00004024 00000000 ff800000 3fff2700  
3fffff40:  3ffecb16 00000000 3fff5954 4023859d  
3fffff50:  cf7ba8c0 00000054 00000000 0000002f  
3fffff60:  00000002 00000018 40228403 3ffee350  
3fffff70:  3ffecaf0 3fffdcc0 3ffeaa08 3ffeaa08  
3fffff80:  40228376 3ffee350 3fffdab0 3fff27c4  
3fffff90:  3fffdc80 00000000 3fff5954 4023131f  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<
```

Sometimes I get a longer stack trace but without the Exception line:
```
Skipping compressed pointer
Skipping compressed pointer

Soft WDT reset

ctx: sys 
sp: 3ffffa60 end: 3fffffb0 offset: 01b0

>>>stack>>>
3ffffc10:  40214bcd 3ffffe20 3fff1fe8 40214acd  
3ffffc20:  78bc4bc5 7c2c9923 ff63a447 ba588908  
3ffffc30:  a3905ed4 dcb919bf ab4bbbe1 fa45e881  
3ffffc40:  49543f72 371709d8 f10baee6 61f75d6e  
3ffffc50:  4000b62e bbf02194 2d50066f 1b16ffe4  
3ffffc60:  4000b6c4 3ffffca0 3ffffe30 ffffffff  
3ffffc70:  00000000 00000000 0000001f 401048fd  
3ffffc80:  4000050c 00000000 4000a333 3ffffda8  
3ffffc90:  4023c0b3 00000030 00000000 ffffffff  
3ffffca0:  40243913 00000001 00000002 40000000  
3ffffcb0:  0000ffff bfffffff 00080240 c0048004  
3ffffcc0:  3ffed7a4 00048000 ff000fff 3ffef9a0  
3ffffcd0:  3fff238c 00000001 3fff277f 00000030  
3ffffce0:  ffffffff 00000000 3ffeaa61 00000008  
3ffffcf0:  40228af2 00000000 3fff238c 00000100  
3ffffd00:  00000050 00000001 40104062 3ffeead8  
3ffffd10:  3ffee080 40100a01 00000000 c0000005  
3ffffd20:  7674785b 3d737265 74710831 6c61746f  
3ffffd30:  7037313d 613d6c64 696c7070 69746163  
3ffffd40:  6f2f6e6f 74657463 7274732d 2c6d6165  
3ffffd50:  67616d69 72752f65 6d692c66 2f656761  
3ffffd60:  2d677770 74736172 6e057265 3d65746f  
3ffffd70:  3d79741d 746f7242 20726568 2d43464d  
3ffffd80:  3237324c 20574430 69726573 70247365  
3ffffd90:  75646f72 283d7463 746f7242 20726568  
3ffffda0:  2d43464d 3237324c 20574430 69726573  
3ffffdb0:  27297365 696d6461 6c72756e 7474683d  
3ffffdc0:  2f2f3a70 43575242 39383330 46443236  
3ffffdd0:  2e413538 61636f6c 0b2f2e6c 6f697270  
3ffffde0:  79746972 0f35373d 5f627375 3d47464d  
3ffffdf0:  746f7242 1a726568 5f627375 3d4c444d  
3ffffe00:  2d43464d 3237324c 20574430 69726573  
3ffffe10:  75197365 435f6273 503d444d 502c4c4a  
3ffffe20:  502c4c43 4c584c43 4652552c 6c6f4307  
3ffffe30:  463d726f 706f4308 3d736569 705f0854  
3ffffe40:  746e6972 46007265 324c2d43 44303237  
3ffffe50:  65732057 73656972 73754300 3d6d6f74  
3ffffe60:  69420854 7972616e 540d543d 736e6172  
3ffffe70:  6572616d 00000001 00000203 00000002  
3ffffe80:  00000008 3233653d 30303834 30382d30  
3ffffe90:  312d6563 2d626431 30303038 3030332d  
3ffffea0:  36633535 33343463 3fff5a66 40214f09  
3ffffeb0:  3fff274c 3fff582c 3fff5a2c 40216e13  
3ffffec0:  40000f58 00000000 00000020 40205f18  
3ffffed0:  3fff274c 3fff582c 3fff5a6c 40237a54  
3ffffee0:  000014e9 00000000 3ffecb24 3fff21bc  
3ffffef0:  000000e0 3fffdcc0 3ffeaa00 4010752c  
3fffff00:  00000000 00000000 3ffeb8f1 3fff21b4  
3fffff10:  3ffecb24 3fff21bc 3fff582c 40239198  
3fffff20:  00000002 3fff274c 3fff274c 3ffef9a0  
3fffff30:  00000000 3fff582c 0000001c 3fff274c  
3fffff40:  3ffecb16 00000000 3fff582c 4023859d  
3fffff50:  b67ba8c0 000004e7 00000000 00000035  
3fffff60:  00000002 00000018 40228403 3ffee300  
3fffff70:  3ffecaf0 3fffdcc0 3ffeaa08 3ffeaa08  
3fffff80:  40228376 3ffee300 00000000 3fff27b4  
3fffff90:  3fffdc80 00000000 3fff582c 4023131f  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000181] 

[000182] [INIT] ESPURNA 1.12.3
```

As I said, I have 4 Espurna devices running. This issue is not limited to the itead-sonoff-touch. I also observed it on Sonoff Basic and Magichome LED controller. I also have the same stability issue when compiling the firmware myself.

When compiling myself, I have access to the the .elf and I used https://github.com/janLo/EspArduinoExceptionDecoder to decode the stack trace : 
```

epc1:     0x30844003
epc2:     0x00000000
epc3:     0x00000000
excvaddr: 0x00000000
depc:     0x00000000

ctx: sys

sp:       0x3ffffd00
end:      0x3fffffb0
offset:   0x000001a0

stack:
0x402396b1: ethernet_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c:1379
0x40228edf: pp_tx_idle_timeout at ??:?
0x40228e52: pp_tx_idle_timeout at ??:?
0x4023131f: system_get_sdk_version at ??:?
```
(Also, I'm not /Users/igrokhotkov, I don't understand why I see this guy's paths)

So is this an issue with LwIP? I have a pretty complex network setup, but I don't understand why it would crash all my Espurna devices.

Any pointers appreciated

xoseperez, 2018-02-08:
Ivan Grokhotkov is one of the main developers of the Arduino Core for ESP8266 project. The devices errors in a pre-built SDK library he compiled, that's why.

You say all four devices reboot within seconds. Could it be something external making them reboot?

Tito1337, 2018-02-08:
I found many stack traces mentionning `etharp.c:1379` in issues reports, most of them pointing to bugs in the ESP8266 Arduino Core library. 

So I decided to upgrade to platform-espressif8266.git v1.6.0, that uses Arduino Core v2.4.0. I now have more than 15 hours of uptime, so I'm hopefull that it fixed it. Sadly using platform v1.6.0 creates slightly bigger firmwares, and the one for the Magichome is now over half a meg...

Possibly related issues in upstream : 
- https://github.com/esp8266/Arduino/issues/1826
- https://github.com/esp8266/Arduino/issues/3555
- https://github.com/esp8266/Arduino/issues/2248
(it is not limited to mDNS though, because I tried with MDNS_SERVER_SUPPORT and MDNS_CLIENT_SUPPORT disabled before upgrading to Arduino Core v2.4.0)

If it stays stable for another two days I will close the issue. Thanks for your quick answer!

xoseperez, 2018-02-09:
Good. Keep us updated.

Tito1337, 2018-02-10:
I can confirm no stability issues with platform-espressif8266 version **1.6.0**. Since it's been fixed in upstream, I close the issue here.

For anybody facing the same issue, I recommend upgrading by changing the platform-espressif version at the beginning of platformio.ini : 
```
platform = https://github.com/platformio/platform-espressif8266.git#v1.6.0
```

closed 2018-02-10 by Tito1337