xoseperez/espurna #539: I have a quiestion..

anarro, 2018-02-08:
Hi,

(I use a Espurna 1.12.3 firmware.)
I like connect 4 buttons (on a wemos D1 mini) with my homeassistant app., I have configured correctly (I think) in a hardware.h file and and enabled "homeassistant" auto discovery button and use a "homeassistant" prefix in a textbox, but when I check in my MQTT.fx app I never view nothing registred on a "homeassistant" prefix... it's correct?
 
[image]

xoseperez, 2018-02-08:
The homeassistant config message is sent on boot or when you enable/disable it. It's only a configuration message, you don't really need it to use ESPurna with HA if you configure it manually.

anarro, 2018-02-08:
Ok Xose, thank you.

closed 2018-02-08 by anarro