xoseperez/espurna #540: no ARP replay after longer runtime

kuppe234, 2018-02-08:
Hello,
after a runtime longer then 1 day (now 1day 20h) I can not connect from any device at the same subnet. Connect from router is possible. I scan ARP packets with tcpdump. But the esp-device give me no arp reply.
If I restart the Wifi (WIFI.RESET) etherthing works, I get an arp replay from esp-device.
I have 2 esp-devices with the same problem.
Do you have a solution?
Greetings, kuppe234.

xoseperez, 2018-02-09:
I have no experience with ARP and ESP8266... searching the web it looks like an old, unresolved (?) issue: https://github.com/esp8266/Arduino/issues/2330, https://github.com/me-no-dev/ESPAsyncWebServer/issues/54.

In the last post, a user suggested this method (`forceARP()`) to be called every second or so:

```
`extern "C" {
    extern char *netif_list;
    uint8_t etharp_request(char *, char *);
}

void forceARP() {
    char *netif = netif_list;
    while (netif) {
        etharp_request((netif), (netif+4));
        netif = ((char *) netif);
    }
}
```

Maybe you can give it a go and report your results?

alex9x, 2018-02-13:
I can reply for this question. I have test 3pcs in one network. 

Firmware I had test: 
- have each 10 seconds - forceARP()
- have ping gateway (192.168.1.1) each 30 seconds

Wifi - WPA or WPA2 - no matter. (WEP I don`t try because it not actual)

Situation: modules power up -> all ok, all 3 have ping and have arp in table on PI server in the same network.
After ~10 minutes - arp without esp :(

root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:39:05   C                     eth0
192.168.1.1              ether   00:0c:42:b7:d4:97   C                     eth0

Then trying to open socket from PHP to 3 devices on 192.168.1.121-122-123. 
First try - socket connection failed - no route to host. In console several times in delay ~1 second:

root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:xx:xx   C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8xx:xx   C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
192.168.1.122                    (incomplete)                              eth0
192.168.1.121            ether   5c:cf:7f:1b:73:ea   C                     eth0
root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:xx:xx   C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
192.168.1.123                    (incomplete)                              eth0
192.168.1.122            ether   5c:cf:7f:1b:79:c0   C                     eth0
192.168.1.121            ether   5c:cf:7f:1b:73:ea   C                     eth0
root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:xx:xx   C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
192.168.1.123                    (incomplete)                              eth0
192.168.1.122            ether   5c:cf:7f:1b:79:c0   C                     eth0
192.168.1.121            ether   5c:cf:7f:1b:73:ea   C                     eth0
root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:xx:xx   C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
192.168.1.123                    (incomplete)                              eth0
192.168.1.122            ether   5c:cf:7f:1b:79:c0   C                     eth0
192.168.1.121            ether   5c:cf:7f:1b:73:ea   C                     eth0
root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:xx:xx  C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
192.168.1.123                    (incomplete)                              eth0
192.168.1.122            ether   5c:cf:7f:1b:79:c0   C                     eth0
192.168.1.121            ether   5c:cf:7f:1b:73:ea   C                     eth0
root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:xx:xx   C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
192.168.1.123                    (incomplete)                              eth0
192.168.1.122            ether   5c:cf:7f:1b:79:c0   C                     eth0
192.168.1.121            ether   5c:cf:7f:1b:73:ea   C                     eth0
root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:xx:xx   C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
192.168.1.123            ether   5c:cf:7f:1a:57:59   C                     eth0
192.168.1.122            ether   5c:cf:7f:1b:79:c0   C                     eth0
192.168.1.121            ether   5c:cf:7f:1b:73:ea   C                     eth0
root@alpha:# arp
Address                  HWtype  HWaddress           Flags Mask            Iface
192.168.1.74             ether   d0:7e:35:f8:xx:xx   C                     eth0
192.168.1.1              ether   00:0c:42:b7:xx:xx   C                     eth0
192.168.1.123            ether   5c:cf:7f:1a:57:59   C                     eth0
192.168.1.122            ether   5c:cf:7f:1b:79:c0   C                     eth0
192.168.1.121            ether   5c:cf:7f:1b:73:ea   C                     eth0
root@alpha:#

In a ~10 seconds after first socket connection try all modules become reachable and works well next ~5 minutes. Then the same unreachable.

All the time router "torch" shows 806(arp) protocol packets from esp with dst 0.0.0.0 
**Also torch shows them with no forceARP() in a code, seems this forceARP only add some more arp packets.**
One time in 40-60 seconds torch shows 4 (802.2) and 88cc packets from esp to 0.0.0.0.

Each 30 seconds torch shows 800(ip) 1(icmp) ping packet from esp to gateway as placed in software.

BUT still become unreachable in a few minutes.

alex9x, 2018-02-16:
New test update:

If each esp (192.168.1.121-123) make pings to pi host (192.168.1.10) - all esp are reachable from pi host. This pings config - 1 packet with 1 sec timeout/interval. 30-50% of pings - failed, but make esp reachable. forceArp are disabled on this tests. The same time - ping try from gateway (192.168.1.1) - failed for 2-4 sequences - then got reply.

gn0st1c, 2018-04-09:
@alex9x : do you still have same issue with the new firmwares?

mcspr, 2019-11-09:
Closing via https://github.com/xoseperez/espurna/pull/1877

closed 2019-11-09 by mcspr