xoseperez/espurna #543: RF bridge with custom firmware support

koenvb, 2018-02-09:
I saw that the new release is supporting new firmware for the Encorder chip on the RF bridge:

> Sonoff RF Bridge support
> * Multiple virtual switches (tested with up to 16)
> * MQTT-to-RF two-way bridge (no need to learn codes)
> * Support for https://github.com/rhx/RF-Bridge-EFM8BB1 custom firmware

Does that mean that whatever code you fill in the GUI, that code will be sent to the chip
as is and the chip will send it out? So more protocols are supported? 

Is there somewhere you could point me to some documentation help how
to flash the firmware of this chip?

Thanks for this project!

xoseperez, 2018-02-09:
Hi
In theory, @rhx or @Portisch might help you with that and on how to flash the firmware on the EFM8, I have not done it myself yet...

rhx, 2018-02-18:
Yes, have a look at the README.md on https://github.com/rhx/RF-Bridge-EFM8BB1
You need to program the custom firmware, which requires opening the case and connecting a programmer (e.g. Arduino) to the relevant headers (similar to what you used to have to do for the ESP8266 before OTA was possible with the default firmware).

Also, while we have been playing with natively learning codes, this isn't working at this stage, but you can use an Arduino with an RF receiver, see https://github.com/pimatic/RFControl

dellk, 2018-02-22:
Hi,
Is it possible to flash back the original firmware? 

Found supposedly a original dump on https://www.letscontrolit.com/forum/viewtopic.php?t=3507 but was unable to flash after converting to hex.

xoseperez, 2018-02-26:
You should ask this question in @Portisch repo: https://github.com/Portisch/RF-Bridge-EFM8BB1

closed 2018-07-26 by stale[bot]