xoseperez/espurna #544: Wifi Reconnect Causes Lights to Flash (Sonoff Relay)

thebradleysanders, 2018-02-09:
It seems when the Wifi connection of the Sonoff Basic running espurna (Latest), causes the relay to turn off the turn back on. This is casing the lights to blink in my house. Is there a code modification that can fix this?

I assume it is in the setup script:
  The Sonoff Recieves Power -> Light On
   Wifi Disconnects casing the Sonoff to reboot - Relay off
   The  Sonoff runs the setup -> Light On

Thanks for you help!

xoseperez, 2018-02-09:
The problem here is that WiFi disconnecting should not reboot the device. When the device reboots there is a brief moment where the relay goes off and once it boots goes back on (if "always on" or "same as before" is selected in the relay boot mode).

So we should focus on why is it rebooting?

thebradleysanders, 2018-02-09:
Yes that seems correct, I have several Sonoff Basics 17 of them have Espurna on them as of yesterday. I have disabled everything except MQTT and alexa support in the UI. I have made no changes to the code, I am running the newest esp8266 core version.

The issue seems to only happen with a few of my sonoffs, as some have an up-time of 21 hours+
Any Recommendations will be highly appreciated, I don't want to switch from Espurna as the interface and features are great!

gn0st1c, 2018-02-09:
can this be your issue?

_Block Interference with a Low Pass Filter - Bust Ghost Switching!_
https://www.youtube.com/watch?v=aq8_os6g13s

thebradleysanders, 2018-02-09:
I don't believe so, I have had the Sonoff installed for a while now, just will a basic MQTT script I wrote.
I recently switched because I changed my Wifi password and it was a pain to reprogram all my Sonoffs.
So I figured while I updated them I would change the firmware to address this issue, so I would not have to do this again. I did not have this issue before Espurna and all my Sonoffs have the same location and hardware configuration (No Modifications, No Sensors). Thanks for the suggestion and input.

thebradleysanders, 2018-02-19:
Any progress on this, it seems there is a similar issue open.
https://github.com/xoseperez/espurna/issues/566

asantos15, 2018-02-25:
@thebradleysanders , have you found any solution to this one yet?
I have tasmota flashed on mine, but it also turns on and off at startup or when the WiFi reconnects. So it seems to be something is checking the pins at startup, causing it to "blink"...

thebradleysanders, 2018-03-02:
No I have not, It seems the problem is related to Wifi signal strength, I replaced my router and it seemed to fix some of my Sonoffs but not all of them. Do yours seem to be affected by distance from your router/AP?

tyjtyj, 2018-06-25:
I have tasmota t1.. The blink too

I about to install sonoff power for air conditioner. I cant afford this to blink

jsponz, 2018-07-14:
Any update on this? I have the same problem on several (not all) devices.

xoseperez, 2018-07-17:
Apparently, it's a problem due to different reasons. I have not been able to reproduce it and other people has reported it happening on different firmware. Improving signal strength seems to help. I'm not sure it's a software problem.

closed 2018-09-22 by stale[bot]

Nightreaver, 2019-05-28:
As for my, ghost switching is gone after chaging the cap.
I currently use a 15nF+4,7nF cap so I'm at 19,7nF currently and reaction sensitivity is still fine.

I'd reccomend to keep this info somewhere in the Wiki.