xoseperez/espurna #545: Silent Touch

giddyhup, 2018-02-09:
I flashed my Sonoff Touch OTA from version 1.10.1 to 1.12.1 and it is now unreachable and the LED isn't flashing. Is there a way to revive it without physical intervention?

xoseperez, 2018-02-09:
Maybe it has gone into safe mode. In that case, it should have created a wifi access point.

giddyhup, 2018-02-09:
Thank you for your quick response. I already scanned for it, there is no AP. : ( I also touched it for at least 10 seconds but I see no difference in behavior. I shut the power off and on again, also no result.

giddyhup, 2018-02-10:
I took it apart and resoldered the pins and it seems the chips is in a crash loop:

```
  ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

```
I can't enter flash mode. Grounding GPIO 0 on boot does nothing. I guess it is bricked. : (

lobradov, 2018-02-10:
If you are reading anything from serial, it’s not bricked :) 
You just have to find gpio0 and ground it before booting. That would put the device into flash mode and you can work with it. You can then erase flash, and re-flash again.

However, this will not likely help. 

See ‘rst cause:1’? That means ‘normal boot’ (in contrast to other options of 2: reset, 3: software crash or 4: watchdog reset). This usually points to bad power as esp understands power on/ off as regular boot cycle..

Still, to be sure, why not leave the device on with serial console for a bit longer and maybe get a bigger picture?

giddyhup, 2018-02-10:
Thanks @lobradov. You are right, as long as there is a heartbeat there is life. This is what I see when I boot with GPIO 0 on ground:
` ets Jan  8 2013,rst cause:1, boot mode:(1,7)`

I am currently considering connecting an esp8266 (wemos d1 mini) and trigger the relay from there. This would actually give me more capabilities. I can connect a PIR and a light sensor to the wemos board (I actually have one setup like that already). I need to figure out what the fourth pin in the connection to the relay board does and if the relay needs to be connected to GND or 3V3 to trigger.

giddyhup, 2018-02-10:
@lobradov your remark regarding power got me thinking. When desoldering the pins the first time I might have messed up some of the traces on the PCB. I am currently connecting GND and VCC to the respective pins on the fixed four-pin header and TX and RX to the respective pads. I'll see if it makes a difference if I connect VCC and GND to the pads.

giddyhup, 2018-02-13:
I couldn't get the module into flash mode again. I gave up.

closed 2018-02-13 by giddyhup