xoseperez/espurna #546: Webserver Schedule Enabled Slide Switch Not Functioning

jnherm, 2018-02-10:
Hello! As I tried to use the Schedule on the Webserver, I end up nothing. Enabled Slide Switch is not functioning.

xoseperez, 2018-02-10:
Hi
There are two issues with the recently added enable switch in the scheduler:
* It always reports as enabled, regardless the actual state
* It doesn't work when you first add a new scheduler, but it does once you reload the page (with an already saved scheduler)

The first one has been addressed in the dev branch (14509efacbe5945f6d0ae233c6773547bd8ad8a0). The last one is still not solved, but you can save and reload the page.

jnherm, 2018-02-10:
Yes i did refresh the page and it not enabled, but schedule is still not working. I will update my firmware with that branch. Thank you so much! More power!

May I ask how can I use MQTT command to set schedule? Also MQTT command to set SWITCHES tab parameters on the webserver?

xoseperez, 2018-02-10:
ESPurna does not have "configuration" MQTT topics. The reason behind this is that if you can send an MQTT command to set a schedule, you can also send an MQTT message to turn on/off the relay a do the scheduling somewhere else.

raghunav, 2018-06-26:
Is there an option to set the schedule using the HTTP API. This will help in creating an Mobile app to on/off lights and also set schedules without the need of MQTT/other controller.

xoseperez, 2018-06-26:
Same answer as before. There is no such feature at the moment. You can still use the HTTP API to switch on/off the relays any time.

xoseperez, 2018-06-26:
Anyway this is out of topic and the original reason for this topic has been long solved. Closing, please check #990.

closed 2018-06-26 by xoseperez