xoseperez/espurna #549: Additional switch doesn't generate "pressed" event

eschava, 2018-02-11:
I've connected EL817 optocoupler (that generates 3.3V when no signal and 0V when signal is present) to the RX pin (GPIO 3) and added -DBUTTON3_PIN=3 -DBUTTON3_MODE="BUTTON_SWITCH+BUTTON_DEFAULT_HIGH" flags to configuration based on instructions from https://github.com/xoseperez/espurna/wiki/Buttons-and-switches

After this, I receive "ESPURNA_XXXXX/button/2" => "2" (BUTTON_EVENT_RELEASED) MQTT message in both cases when there is 3.3V and 0V on the input

Also, I think there is a typo in WIKI page because -DBUTTON_MODE (w/o 3) is used for configuring BUTTON3 and also "|" char cannot be used in platformio.ini. I got error below if pipe is used:

Compiling .pioenvs/itead-sonoff-basic-dht-ota/src/espurna.ino.o
sh: BUTTON_DEFAULT_HIGH: command not found
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs/itead-sonoff-basic-dht-ota/src/pwm.o
Archiving .pioenvs/itead-sonoff-basic-dht-ota/libFrameworkArduinoVariant.a
sh: BUTTON_DEFAULT_HIGH: command not found
Indexing .pioenvs/itead-sonoff-basic-dht-ota/libFrameworkArduinoVariant.a
*** [.pioenvs/itead-sonoff-basic-dht-ota/src/espurna.ino.o] Error 127
*** [.pioenvs/itead-sonoff-basic-dht-ota/src/pwm.o] Error 127
Compiling .pioenvs/itead-sonoff-basic-dht-ota/FrameworkArduino/Esp.o
sh: BUTTON_DEFAULT_HIGH: command not found
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
*** [.pioenvs/itead-sonoff-basic-dht-ota/FrameworkArduino/Esp.o] Error 127
=========================== [ERROR] Took 2.36 seconds =======================

xoseperez, 2018-02-11:
If you have configured the button as BUTTON_SWITCH it only reports BUTTON_EVENT_CLICK (code 2 also). If you want to know when it was pressed and released configure it as BUTTON_PUSHBUTTON.

I have fixed the documentation in the wiki. Thank you!

eschava, 2018-02-12:
Thanks, it works now!
But docs still don't look clear. It states that:

> BUTTON_PUSHBUTTON - connected button is of push-button type and event is fired when button is released.

xoseperez, 2018-02-12:
I have updated the wiki again so it's more clear.
Thank you!

closed 2018-02-12 by xoseperez

eschava, 2018-02-12:
Thanks a lot for your work!