xoseperez/espurna #551: Sonoff 4CH Pro AP mode

jeanpaul79, 2018-02-11:
Hello,

I'm using a Sonoff 4CH Pro with 1.12.3.

I've commented `#define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
#define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE`
and when I do a long press on button 1, it goes to AP mode but after 2-3s, it connects back to wifi.

Do you know how to use AP mode with this sonoff 4CH pro please.

Best regards,

xoseperez, 2018-02-12:
I should check this, it should stay in AP mode for at least 2 minutes if I recall.
I'm on holidays now, will give it a look next week.

xoseperez, 2018-03-02:
Can't reproduce this issue without more info. Can you copy here the debug log while doing the before procedure?

jeanpaul79, 2018-03-03:
I'm using 1.12.3 and I modified code/espurna/config/hardware.h and I commented line 484 & 485 #define BUTTON1_LNGCLICK BUTTON_MODE_NONE & #define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE

With Sonoff 4CH Pro connected to my wifi, I do a longpress on button 1 and it seems to turn to AP mode (according to LED) but then it connects to wifi immediatly.

I'll disconnect it from main power and do a debug log.

jeanpaul79, 2018-03-03:
Hello,

Here is the log:
[putty.log](https://github.com/xoseperez/espurna/files/1777491/putty.log)

xoseperez, 2018-03-04:
In the log, you can see a "Button #0 event 4" that is resetting the device. Event 4 is BUTTON_EVENT_LNGCLICK. When you commented the 484 line the firmware uses the default action for a long click on button 0 defined in defaults.h. And that's a reset.
The defaults for the first hardware button are double click for AP mode, long click (>2s) for reset and long-long click (>10s) for factory reset.

jeanpaul79, 2018-03-04:
Oh ok thanks. Yes according to what is written in hardware.h I thought that a long press switch to AP mode.

I've replaced the line as you said.

Thanks.

xoseperez, 2018-03-05:
Staged for release

closed 2018-03-05 by xoseperez