xoseperez/espurna #552: Switch logic gets inverted

Tardymo, 2018-02-11:
If there is a switch tied to relay and relay toggling occurs (button press, MQTT, web e.t.c) while switch is "on", that switch gets inverted so that turning it "off" actually switched relay "on". Is there an easy way to fix this?
There are two different ways it could work and I'll give an example using SONOFF_RF
Turn relay on using swith connected to GPIO14 and then toggle relay using push button or RF remote. At this point relay is off and turning swith off will actually turn relay on. But instead more logical way would be to leave relay off when switch is turned back off. Or second option would be to not allow turning relay off while switch is turned "on".

xoseperez, 2018-02-12:
You can configure your switch as a pushbutton with explicit action for press and release. You will also want to disable any secondary options on this button to avoid unwanted side-effects (ap mode, factory reset). This could be a starting point:

```
    #define BUTTON1_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON1_PRESS           BUTTON_MODE_ON
    #define BUTTON1_CLICK           BUTTON_MODE_OFF
    #define BUTTON1_DBLCLICK        BUTTON_MODE_NONE
    #define BUTTON1_LNGCLICK        BUTTON_MODE_NONE
    #define BUTTON1_LNGLNGCLICK     BUTTON_MODE_NONE
```

Change BUTTON_MODE flags to match your hardware (maybe it's default low or it already has a pullup).

Tardymo, 2018-02-17:
Actually I need both buttons to act independently of one another and swith the same single relay. So one button should be a toggle button and the other should be an on/off switch. If at least one of the buttons in in "on" position, relay should be switched on. For example if first button is clicked, relay switches on and no matter what I do with the second button, relay stays on until button is clicked again. And vice versa - if second button (switch) is pressed - relay stays on until button is released. Toggling the first button should not swith the relay off. However ralay should stay on if second button is released and first button was clicked once.
Think of both buttons as being connected to OR gate output of which controls the relay.

As far as I know, current firmware will not allow such configuration. But as I remember, there were plans to make buttons and relays independently configurable, so if two buttons are controlling one relay, nice feature to have would be to define logical function between buttons (AND, OR, XOR).

Tardymo, 2018-03-04:
I patched the code to make it work the way I need until such functionality is available in stock espurna firmware, but there is one issue. Here's what I did:
```
// Buttons
#define BUTTON1_PIN             0
#define BUTTON1_MODE            BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
#define BUTTON1_RELAY           2
#define BUTTON2_PIN             14
#define BUTTON2_MODE            BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
#define BUTTON2_PRESS           BUTTON_MODE_ON
#define BUTTON2_CLICK           BUTTON_MODE_OFF
#define BUTTON2_DBLCLICK        BUTTON_MODE_NONE
#define BUTTON2_LNGCLICK        BUTTON_MODE_NONE
#define BUTTON2_LNGLNGCLICK     BUTTON_MODE_NONE
#define BUTTON2_RELAY           3

// Relays
#define RELAY1_PIN          12
#define RELAY1_TYPE         RELAY_TYPE_NORMAL
#define RELAY2_PIN          100
#define RELAY2_TYPE         RELAY_TYPE_NORMAL
#define RELAY3_PIN          100
#define RELAY3_TYPE         RELAY_TYPE_NORMAL
```
And then in relay.ino at the end of relayLoop function I added this line:

`_relays[0].target_status = _relays[1].current_status || _relays[2].current_status;`

As you can see, I created two dummy relays which are assigned to each of the buttons and then actual relay is controlled by OR-ing two dummy relays.
The problem is I need the second dummy relay to switch off when button 2 is released, but that does not happen. In debug output I see this:
```
[020199] [BUTTON] Button #1 event 1
[020201] [RELAY] #2 scheduled ON in 0 ms
[020214] [RELAY] #2 set to ON
[020226] [RELAY] #0 set to ON
[021227] [RELAY] Saving mask: 5
[021408] [BUTTON] Button #1 event 4
```
BUTTON_MODE_OFF does not work for some reason after click event
I can make it work by defining button 2 as BUTTON_SWITCH, but if for some reason dummy relay 2 gets toggled by MQTT or REST, button functionality will be inverted and I must prevent that from happening.

xoseperez, 2018-03-05:
I think part of the problem is the naming. BUTTON_PUSHBUTTON is a momentary push button: you press it, you might hold it down and you released and it goes back to the original state. A BUTTON_SWITCH is a wall switch, ever time you click it (or slide it, or toggle it) it toggles between teo stable states.

Now, there is a tricky thing in ESPurna with indexes. All lists (arrays) are 0-based, so the "Button #1 event 1" log line above is actually BUTTON2_PRESS and the "Button #1 event 4" is BUTTON2_LNGCLICK, that you have defined as "do nothing".

Tardymo, 2018-03-05:
Yes, PRESS event is fine, but I am not getting CLICK event after button is released. Instead there is "event 4". I am releasimg the button quickly so LNGCLICK should not have been triggered. Are LNGCLICK and LNGLNGCLICK triggered on button release? In that case these events must be set to BUTTON_MODE_OFF in my case, right?

xoseperez, 2018-07-01:
Closing since there has been no feedback for several months. Feel free to reopen if necessary.

closed 2018-07-01 by xoseperez