xoseperez/espurna #553: Espurna not working with IOBroker Mqqt Server

Alveran81, 2018-02-12:
It's a long thread. Can somebody link the orginal?

https://github.com/ioBroker/ioBroker.mqtt/issues/13#issuecomment-364797655

xoseperez, 2018-02-12:
I'm on holidays and don't have any device with me. Will test it next week.

Alveran81, 2018-02-14:
OK. Sorry it works with this settings. 
[image]
[image]
But how to set Channel and relay?

xoseperez, 2018-02-14:
Using:

```
LEDLicht/relay/0/set
LEDLicht/channel/0/set
LEDLicht/channel/1/set
LEDLicht/channel/2/set
LEDLicht/channel/3/set
```

Basically: MQTT commands end with `/set`.

Alveran81, 2018-02-15:
Yes i know. I think its a Problem in IObroker.

xoseperez, 2018-07-01:
Closing this since there has been no activity or feedback for months. Please, feel free to reopen if necessary.

closed 2018-07-01 by xoseperez

hurzhurz, 2018-08-27:
I'm starting to play a little bit with ioBroker and stumbled across this problem.
So the issue is that ioBroker automatically creates object-ids for e.g. the brightness-set-topic, but not for the set-topics of channels and relays?

I think the reason is that it learns them by looking on what topics are subscribed.
And as espurna subscribes the own setter topics, the auto-creation works fine for brightness:
"ESPURNA-XXXXX/brightness/set"
But not for channels/relays because of the wildcard:
"ESPURNA-XXXXX/channel/+/set"

What about doing a publish for every setter topic with an empty message?
Maybe after the connect to the server when the subscriptions are made too.

This should also work when the MQTT adapter of ioBroker is used as client instead of server.
I have seen tasmota doing something similar for e.g. "cmnd/sonoff/POWER" (but I don’t know if it does that because of the same purpose).