xoseperez/espurna #557: Espurna as a generic ESP framework, without a switch and a LED, thus freeing two GPIO

PieBru, 2018-02-12:
Hi, although Espurna is born on the wave of ESP switches, IMO it differs radically from other frameworks as it can be also a good generic ESP firmware that can be adapted to lots of solutions very different from wifi switches.

With this goal in mind, I'm trying Espurna on bare bone ESP-12F and Wemos D1 mini without a relay and without a predefined LED behaviour, but as of 1.12.4a by defining ESPURNA_CORE the switch#0 is included in the UI and the LED doesn't have an "unhandled" option, therefore occupying two precious GPIO.

As a generic firmware, ESPURNA_CORE should not include switch#0 and the LED should have a "Custom" choice in the drop-down. This should be limited to ESPURNA_CORE without any impact the currently supported switches.
 
Piero

PieBru, 2018-02-14:
Solved by removing -DITEAD_SONOFF_BASIC from the ESPURNA_CORE env definition.
I added also ota envs to core 1m and 4m like this:
```
[env:espurna-core-4MB-ota]
platform = ${common.platform}
framework = arduino
board = d1_mini
board_flash_mode = dout
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags} -DESPURNA_CORE -DMANUFACTURER=\"ESP\" -DDEVICE=\"CORE\"
upload_speed = 115200
upload_port = "192.168.4.1"
upload_flags = --auth=fibonacci --port 8266
monitor_baud = 115200
extra_scripts = ${common.extra_scripts}
```

xoseperez, 2018-03-02:
I have defined the ESPURNA_CORE image as a special device in `hardware.h`.

xoseperez, 2018-03-02:
Staged for release. Thanks @PieBru .

closed 2018-03-02 by xoseperez