xoseperez/espurna #559: Espurna + Blynk server

artemicruz, 2018-02-13:
It would be wonderful to add blynk server connection.
add blynk libraries are open source
the user only needs: server, port and tocken (online or private) 

http://docs.blynk.cc/#hardware-set-ups-esp8266-standalone

thaeger71, 2018-09-19:
Is someone working on this this ? Otherwise I would start to work on it .... but just for case there is already something, I would offer my help.

xoseperez, 2018-09-19:
There is no active development on this, at least not in the main repo.

thaeger71, 2018-09-26:
Hey guys,

I forked the master branch and implented Blynk support. How is the correct way to get it into the regular branch ? Do I have to make a pull request ?

xoseperez, 2018-10-04:
The code looks mostly OK. I have not tested it yet. But I would not enable Blynk by default unless someone points out a very good reason to do it.

bluexp1, 2019-02-16:
Is there any timeline to release this feature?

thaeger71, 2019-02-17:
It's a good question, I mean Max made also some effort to optimze things, now everthing seems to work, but so far I know there is no feedback from Xose on this so far ....

mcspr, 2019-02-21:
Yep. #1230 PR is still pending (+ maybe needs another thought about wiring more things... and how that wiring is happening)
I have not used it very much though.

thaeger71, 2019-02-21:
I marked everything pending as solved regarding #1230 is there something I have to in addition from my side ?

xoseperez, 2019-02-21:
I'm going to release 1.13.4 first and then merge the code into `dev`.