xoseperez/espurna #560: Schedule not working on Sonoff 4Ch

Majwoj, 2018-02-13:
Hi
I'm using a schedule in Domoticz, and it works in Domoticz, and sometimes works on Sonoff with ESPURNA.
I've noticed that it not always turn on switches.

xoseperez, 2018-02-13:
Sorry, I don't know how the Domoticz Schedule works. I guess it sends an MQTT message when the time arrives. Can you check if the message is received by your ESPurna device?

Majwoj, 2018-02-13:
I’m newbie in ESPURNA 😊 where can I find logs?

xoseperez, 2018-02-13:
You can do it wired (like when you first flash it) opening a serial port to it at 115200 bauds or via telnet connecting to the IP of the device (you might need to first enable this option in the Admin tab.

Majwoj, 2018-02-13:
I’ve used telnet.

I’ve scheduled in Domoticz „off” on 20:25 and it works, and „on” on 20:30 – it doesn’t work.

I’ve attached log file…..

xoseperez, 2018-02-13:
Sorry, I cannot see the attached log file. Maybe it doesn't work from email...