xoseperez/espurna #561: Unexplained behaviour of Sonoff RF Bridge when using car alarm fob.

mr-sneezy, 2018-02-13:
I tried to find a help forum to discuss ESPurna functionality elsewhere, but in the end it seems this is it ?

Question rather than issue... 
I use my Espurna flashed Sonoff RF Bridge to lock my car at night by the cars remote alarm & central locking system, in case I forget to do it...
 This works fine via a simple time trigger automation from HomeAssistant sent to the bridge. It can unlock it too by a second learned code, but I don't need that right now.

The odd behaviour is that if I lock my car now with my car alarm key fob, the car locks multiple times in a few seconds. I think maybe Espurna detects the RF code from the bridge and then 'repeats' it multiple times...
Is this coming from the firmware, if so how can I stop this repeating of my key fob signal ?
Thanks.

albacete99, 2018-02-13:
I have same issue. I have a alarm with rf codes to lock or unlock.
When I send RF code (previusly learning in ESPurna Web) in HomeAssitant automation or Espurna Web, It repeats multiple times.
I tried to learn different remote but when unlock alarm It repeats 2/3 times Rf code
Thanks

xoseperez, 2018-02-13:
ESPurna repeats any code received via MQTT 4 times (that's the value of the RF_SEND_TIMES setting in general.h). But I agree it should not do it when the message comes via RF already. Will fix this.

Tardymo, 2018-02-13:
Do you have some custom Rf receiver connected to your car's central locking? Stock remotes have rolling codes so adding fixed code could be a security concern as somebody could sniff that fixed code and unlock your car...

xoseperez, 2018-02-13:
@albacete99  @mr-sneezy  Can you test the latest version from the dev branch?

mr-sneezy, 2018-02-14:
@albacete99 - Yes I will try it out. I used the binary to load the RF Bridge so I may have to set up the compiling environment first.
@Tardymo - Yes, I assumed the same as the aftermarket system is only a couple of years old, but I tried the key fob learning anyway and it was quickly revealed that it's not a rolling code...

mr-sneezy, 2018-02-14:
I had a go at setting up VS Code and PlatformIO to compile the Dev code. I had some problems with Platform IO and updating it, so I've stopped for tonight. I'll try again tomorrow night, but if there is a possibility to build us a Dev bin to test please let us know. 
Thanks,
Martin

albacete99, 2018-02-14:
Hi @xoseperez. I tested lastest version from dev (my first compiling with ArduinoIDE) and flash via Web (no serial).
But this version doesn't work for me. When I switch on to send RF command, alarms receives 2 codes (2 beeps).
Thanks

mr-sneezy, 2018-02-15:
@xoseperez
Hi, good news. It took me two nights work to install and tune the IDE, I eventually gave up on VS Code with PlatformIO as it had too many confusing error messages, installed Arduino portable and many libraries and compiled Dev OK.
Bug is fixed. Thanks. Car now locks only once with key fob but still works with my HA automation fine.
Martin

mr-sneezy, 2018-02-15:
Oh oh, I spoke too soon. That repeating bug seems fixed but now I have random unavailability in HomeAssistant and I get a flood of door and door bell events with it when it recovers... Will need to shut the RF bridge down till I can work out why tomorrow night.

albacete99, 2018-02-15:
Hi again. I've just read @mr-sneezy  problem with HomaAssistant and Dev version and I had same problems but I thought it was a problem because of HA but the same happened to me, sensors via mqqt that became unstable.
I flashed lastest stable version and it works perfect

mr-sneezy, 2018-02-16:
OK I did a full erase of the ESP8285 via FlashESP8266.exe and a 1Mb blank file, then retried the DEV build again tonight and reflashed it, and I still have issue with loss of availability in HA and false events as it drops in and out. I'll go back to master until we hear more.

mr-sneezy, 2018-02-17:
Tonight I read this issue #572 and it seemed to me the issue was similar, so I tried the same fix with two changes to the Arduino IDE board settings. Changed to Generic ESP8285 Module and changed IwIP Variant to V1.4 Prebuilt. 
So far my complied version of Dev with this change is stable and the repeater bug is also gone.
If it goes unstable I'll report here again...

mr-sneezy, 2018-03-01:
Not sure if this is relevant to this bug or it's new, but my unit is now playing up every few minutes with 'unavailable' reported by Home Assistant GUI. Each time it becomes available again I get a flood of reports from RF switches monitored by the bridge. Very weird after so many days working fine...

xoseperez, 2018-03-04:
Do you mean MQTT reports of messages from the RF switches? The RFBridge does not enqueue incoming RF packets.

mr-sneezy, 2018-03-04:
Hi xoseperez. I'm not sure I know enough to answer that question with confidence. I think HA tries to set the state of RF switches when the RFBridge becomes available, and that's what is cyclically setting off my door bell and car locks. I have a suspicion that the RFBridge is resetting now every few minutes but I don't know why (I have not powered it on for a few days now since the trouble still occurred during the middle of the night...).
Is there a way for me to tell you better from any logs ?

albacete99, 2018-03-04:
Hi again mr-sneezy.
Try to disconnect nmap Discovery un HA. I had the same problem but It stopped when i did It.

xoseperez, 2018-03-05:
@mr-sneezy  The best way would be to have the logs of the device when that happens. Since the problem seems to be related to connectivity and/or resetting the device you should wire it to your computer using a USB2UART device, temporarily disable the communications with the secondary microcontroller (remove the jumpers) and enable DEBUG_SERIAL_SUPPORT from the hardware.h file.