xoseperez/espurna #562: Dimmer Scheduler

wysiwyng, 2018-02-13:
Is support for scheduled dimming already planned? I want to use Espurna on a NodeMCU board with multiple PWM controlled LED strips, and be able to change their brightness via a schedule. This might also be doable with MQTT, but I prefer to use this otherwise great software in a standalone fashion.

I had a look at the code, and it seems that firmware-side implementation should be not too difficult. However I have only rudimentary knowledge of web UI programming and thus no clue how to start there.

Tardymo, 2018-02-13:
Do you have a special Rf receiver connected to your car's central locking? Stock remotes have rolling codes so adding fixed code could be a security concern as somebody could sniff that fixed code and unlock your car...

xoseperez, 2018-02-13:
@wysiwyng  I'm flagging this as an enhancement but I don't plan to add such functionality soon...

@Tardymo  I guess this comment goes to #561 .

Tardymo, 2018-02-13:
Yep, somehow misposted on my phone. Deleted the comment.

wysiwyng, 2018-03-21:
solved in #574.

closed 2018-03-21 by wysiwyng