xoseperez/espurna #563: Echo 2 plus can't find ESPurna

ManuelW77, 2018-02-13:
Hey there,
i have a ESP8266-12F with the latest ESPurna running. I have it compiled with the
Magic Home V2.3 option.

Everything works well, but Alexa can't find the ESPurna. I try'd it several times, rebooted the ESP, but nothing works. Alexa did not find the ESP.

xoseperez, 2018-02-13:
ESPurna uses (my own) fauxmoESP library. Some people has reported issues when using it with v2 devices (like Echo 2 Plus) when compiled with Arduino Core v2.4.0. Try compile it using Arduino Core 2.3.0.

https://bitbucket.org/xoseperez/fauxmoesp/issues/39/generation-2-devices-support

ManuelW77, 2018-02-13:
Thanks it solved the problem. With Core 2.3.0 it works fine. :)

xoseperez, 2018-02-13:
Good. I'm closing this issue since it should be fixed in the fauxmoESP library issue above.

closed 2018-02-13 by xoseperez