xoseperez/espurna #564: DebouncEvent Library Compile Issue

shredder87, 2018-02-13:
HI,

Having some issues with the latest espurna and DebouncEvents when building for ITEAD_SONOFF_BASIC.

I am guessing I have the wrong version of something. I have tried the two DebouncEvents libraries attached and they both have the same errors.
[xoseperez-debounceevent-e53c98caf8f3.zip](https://github.com/xoseperez/espurna/files/1720628/xoseperez-debounceevent-e53c98caf8f3.zip)
[xoseperez-debounceevent-a5c041ec8a6a.zip](https://github.com/xoseperez/espurna/files/1720629/xoseperez-debounceevent-a5c041ec8a6a.zip)

Compiler output:

Arduino: 1.8.5 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

C:\sandbox\Sonoff_Wifi_Switch\espurna-master\code\espurna\button.ino: In function 'void buttonSetup()':

button:137: error: invalid conversion from 'int' to 'callback_t {aka void (*)(unsigned char, unsigned char)}' [-fpermissive]
             _buttons.push_back({new DebounceEvent(BUTTON1_PIN, BUTTON1_MODE, BUTTON_DEBOUNCE_DELAY, btnDelay), actions, BUTTON1_RELAY});
                                                                                                             ^

In file included from C:\sandbox\Sonoff_Wifi_Switch\espurna-master\code\espurna\button.ino:13:0:

C:\Program Files (x86)\Arduino\libraries\xoseperez-debounceevent-a5c041ec8a6a\src/DebounceEvent.h:54:9: error:   initializing argument 2 of 'DebounceEvent::DebounceEvent(uint8_t, callback_t, uint8_t, long unsigned int)' [-fpermissive]
         DebounceEvent(uint8_t pin, callback_t callback, uint8_t defaultStatus = HIGH, unsigned long delay = DEBOUNCE_DELAY);
         ^

button:143: error: invalid conversion from 'int' to 'callback_t {aka void (*)(unsigned char, unsigned char)}' [-fpermissive]
             _buttons.push_back({new DebounceEvent(BUTTON2_PIN, BUTTON2_MODE, BUTTON_DEBOUNCE_DELAY, btnDelay), actions, BUTTON2_RELAY});
                                                                                                             ^

In file included from C:\sandbox\Sonoff_Wifi_Switch\espurna-master\code\espurna\button.ino:13:0:

C:\Program Files (x86)\Arduino\libraries\xoseperez-debounceevent-a5c041ec8a6a\src/DebounceEvent.h:54:9: error:   initializing argument 2 of 'DebounceEvent::DebounceEvent(uint8_t, callback_t, uint8_t, long unsigned int)' [-fpermissive]
         DebounceEvent(uint8_t pin, callback_t callback, uint8_t defaultStatus = HIGH, unsigned long delay = DEBOUNCE_DELAY);
         ^

C:\sandbox\Sonoff_Wifi_Switch\espurna-master\code\espurna\button.ino: In function 'void buttonLoop()':

button:238: error: 'class DebounceEvent' has no member named 'getEventCount'
                 unsigned char count = _buttons[i].button->getEventCount();
                                                           ^

button:239: error: 'class DebounceEvent' has no member named 'getEventLength'
                 unsigned long length = _buttons[i].button->getEventLength();
                                                            ^

exit status 1
invalid conversion from 'int' to 'callback_t {aka void (*)(unsigned char, unsigned char)}' `[-fpermissive]

xoseperez, 2018-02-13:
I build ESPurna against DebounceEvent version tagged 2.0.1, right now it's the tip of the master branch.

xoseperez, 2018-02-26:
@shredder87  Have you tested upgrading the library? Did it work?