xoseperez/espurna #565: H801 per channel on/off

Anroxo, 2018-02-13:
Hi!
I tried it with 6 dummy relays and it didn't work. It is okay with 5. For me the it would be best if one relay would switch the RGB channel and two separate  ones the two white channels. Can someone confirm that it does work with 6. Or should I consider it as a hack and wait for better support?

xoseperez, 2018-02-13:
Right now ESPurna only supports up to 5 dimmer channels but adding more it's a matter of hacking the lightSetup method in light.ino (I think). 

To achieve what you say about having different dummy relays controlling different channels could be achieved using a mapping (like with buttons and relays). Right now the implementation is much simpler (one switch for every channel or one switch for all channels).

Anroxo, 2018-02-13:
I see. Thanks for the information.

ColinShorts, 2018-05-13:
@xoseperez as per #830 should this now work as per the comments in relay.ino under RELAY_PROVIDER_LIGHT - dummy relays set to 5 gives a per channel switch, and set to 6 gives a master plus one for each channel?

xoseperez, 2018-05-15:
Right.

ColinShorts, 2018-05-15:
Thanks @xoseperez, I've not got around to checking this yet, @anroxo can you test and confirm?

xoseperez, 2018-05-18:
Will be released soon

closed 2018-05-18 by xoseperez