xoseperez/espurna #566: Trouble staying on wifi

lmarlow, 2018-02-13:
My Wemos board has trouble staying online. If I reset it, it will usually reconnect. However, it seems to be offline the majority of the day. It's installed in my garage with a relay shield (GPIO 5), DHT22 (GPIO 14), and reed switch on GPIO 16. I'm not sure how to go about debugging why it loses its connection. RSSI seems to be stay between -50 and -60 dBm. Is there a way to log debug info to the board and download it later when it's back online or by connecting it directly through USB?

I noticed that `WEMOS_D1_MINI_RELAYSHIELD` defines to `I2C_SCL_PIN` to GPIO 14, the same as the DHT22. Could this cause issues?

Here is the last few days of connectivity according to my Unifi controller:

DATE/TIME | DURATION | DOWN | UP
--|--|--|--
02/13/2018 8:59 am | 1h 30m 38s | 991 KB | 18.6 MB
02/11/2018 8:30 pm | 12h 53m 50s | 991 KB | 18.6 MB
02/10/2018 11:47 am | 8m 34s | 6.85 KB | 107 KB
02/09/2018 1:39 pm | 3h 24m 25s | 297 KB | 5.1 MB
02/08/2018 11:09 pm | 9h 37m 38s | 714 KB | 13.5 MB
02/08/2018 7:07 pm | 6m 26s | 3.33 KB | 42.5 KB

Configuration|Value
--|--
Manufacturer|WEMOS
Device|D1_MINI_RELAYSHIELD
Channel|11
RSSI|-60 (6m)
Firmware name|ESPURNA
Firmware version|1.12.3
Firmware build|2018-02-02 09:58:18
Current time|2018-02-13 10:30:15
Uptime|0d 01h 31m 44s
Free heap|22576 bytes
Firmware size|477728 bytes
Free space|2666496 bytes
MQTT Status|CONNECTED

xoseperez, 2018-02-13:
I2C is not enabled if no I2C sensor is defined, so this cannot be the issue. There is no way to store the log locally atm. Can you monitor the MQTT messages it sends to see if it's rebooting also?

lmarlow, 2018-02-13:
It does look like it has rebooted when it comes back online. I will track uptime via mosquitto_sub to verify.

xoseperez, 2018-02-13:
If it is rebooting (for whatever reason) more than 5 times in a row (less than 60 seconds each) it will go into "safe mode" which means it will not try to connect to the WiFi networks defined but it will create an AP.

lmarlow, 2018-02-20:
I haven't noticed any particular patterns yet. It doesn't seem to correlate to temperature nor Vcc. When it is offline for extended periods, I don't see it creating it's AP. I have 2 Sonoff basics that stay connected.

[image]

xoseperez, 2018-02-26:
I don't see any reason for this. If you log into it via telnet and write `crash` it should dump the latest stack dump, then you can use the Exception Decoder to see where the error happened (https://github.com/me-no-dev/EspExceptionDecoder).
If it doesn't work, I would wire the device to the computer via USB and monitor it until it fails...

lmarlow, 2018-02-26:
Ok... I will give that a shot.

lmarlow, 2018-02-28:
It's been up for > 13 hours now. I telnet in and `CRASH` only reported:

```
[167137] [DEBUG] No crash info
[167138] +OK
```

I will try to get platformio setup on my linux server and log the serial console from there.

mr-sneezy, 2018-03-02:
I've had similar issue with ESP8266 modules just like that in the past, and it turned out to be noise on the 5V usb power supply from the wall wart converter I was using. ESP8266 chips need a good clean source of 3.3V, and if the 3.3V regulator or the 5V source above it is not well filtered they are very flakey with staying connected. 
Have you tired a different power source already ?

lmarlow, 2018-03-06:
I connected it to a computer's USB for power and to log the terminal output. It stayed online for 4 straight days. Unfortunately, multiple things changed when doing that: different power supply, closer to my wifi AP, garage door detector and relay disconnected. I guess my next step is to use the same power supply as before, but where it is now to see if it gets the problem again.

Is there a good way to measure the quality of the power supply or the regulator?

mr-sneezy, 2018-03-07:
Not without an oscilloscope that can go to about 1Mhz looking for fast noise or short voltage changes above the norm.
If not then try a good 5V USB wall wart brand that claims 2Amps current delivery. Even though the ESP8266 averages 80mA when communicating it has much higher peak demands.

lmarlow, 2018-03-15:
I seem to have narrowed it down to it only having issues when the reed switch is connected. I'm going to take yet another look at my soldering and layout, maybe try it with a shorter wire. Right now the switch is connected via a twisted pair of a 30 foot cat 5 cable.

I'll close this as it is most likely an issue with my side of things. Thanks for the help.

closed 2018-03-15 by lmarlow

mr-sneezy, 2018-03-16:
30 feet of antenna connected to a high impedance 3.3V pin on the MCU :)
Put a strong pull down to ground and a decoupling cap on the pin...

lmarlow, 2018-03-16:
@mr-sneezy, I was wondering if something like that was going on. I'll give those a shot. I'm a software developer by trade and just getting started dabbling in electronics/home automation: I'm not used to physics playing much of a role in solutions.

Thanks 🎉

mr-sneezy, 2018-03-17:
This might be relevant. If the problem is only minor then a 0.1uF cap and a resistor might be enough, if not then they expand on what you can do. There are two variations depending on if your switch or reed is connected to GND or VDD on the common side.
https://www.digikey.com/en/articles/techzone/2012/apr/protecting-inputs-in-digital-electronics

lmarlow, 2018-03-26:
@mr-sneezy, I added a 1k resistor between the switch and the input pin to protect against too much current and a 0.1uF cap from the pin to ground for voltage ringing. It's been continuously online since. Thank you very much for the tip and link to the digikey article. It seems to be exactly what I needed.