xoseperez/espurna #567: Telnet disconnect

AlbertWeterings, 2018-02-14:
I found again strange behavior on my self compiled firmware, all precompiled files don't have this issue. Many functionality is disabled (see below). I use tenet for debugging my observation is telnet being disconnected after around 1 hour (telnet timeout message). The unit gets every 5 minutes some data over UART that is send to a MQTT broker. After disabling line 108 in telnet.ino the connection is not closed anymore and messages "[TELNET] Timeout on client #0 at xxx" is passing by intermittently.

Is the timeout to short or is there something else going on?

#define ALEXA_SUPPORT          0
//#define DEBUG_SERIAL_SUPPORT   0
//#define DEBUG_TELNET_SUPPORT   0
//#define DEBUG_UDP_SUPPORT      1
#define DOMOTICZ_SUPPORT       0
#define HOMEASSISTANT_SUPPORT  0
//#define I2C_SUPPORT            1
//#define INFLUXDB_SUPPORT       1
//#define IR_SUPPORT             1
//#define LLMNR_SUPPORT          1  // Only with Arduino Core 2.4.0
//#define MDNS_SERVER_SUPPORT    0
//#define MDNS_CLIENT_SUPPORT    1
//#define BROKER_SUPPORT         0
//#define MQTT_SUPPORT           0
//#define NETBIOS_SUPPORT        1  // Only with Arduino Core 2.4.0
//#define NOFUSS_SUPPORT         1
//#define NTP_SUPPORT            0
//#define RF_SUPPORT             1
#define SCHEDULER_SUPPORT      0
//#define SPIFFS_SUPPORT         1
//#define SSDP_SUPPORT           1
//#define TELNET_SUPPORT         0
//#define TERMINAL_SUPPORT       0
#define THINGSPEAK_SUPPORT     0
//#define WEB_SUPPORT            0

xoseperez, 2018-02-14:
That line just closes the TCP client when it has timed out to free resources. If you remove the line the client will not close properly and you won't be able to reconnect to it. If you don't want the client to time out you should send something every few minutes.

AlbertWeterings, 2018-02-15:
Thanks Xose, I understand the process of timing out. The thing is this happens intermittently (with changing intervals) I would expect a fixed interval. I disabled the line to prevent is from closing the session so I could monitor intervals as still the debug message is coming by now.

And what is also strange is that only the versions I compiled are having this behavior the pre-compiled versions from https://github.com/xoseperez/espurna/releases keep going for days without me interacting.

In theory I shout never have to send something from the client to the server as the client already send an acknowledgement for every received message.

xoseperez, 2018-02-16:
It would be interesting to know what are the differences between your build and mines. You can check the exact versions (and revisions) of the libraries I use in the platformio.ini file in the `code` folder. Also, I compile them against Arduino Core 2.3.0.

AlbertWeterings, 2018-02-17:
I think libraries are all the same. Found one big difference I compile them with esp8266 Community version 2.4.0.

I also found the issue https://github.com/xoseperez/espurna/issues/572 reported by gd-99 witch is  already closed solves more of my problems after setting IwIP variant to V1.4 Prebuilt my MQTT disconnect issue https://github.com/xoseperez/espurna/issues/436 is also solved.

The telnet intermittently disconnecting is not yet solved, but I observed that it happens directly after a DEBUG message is partial received. First I see the [{time}] than it stops for about 1 second followed by the message and right after that is closes the connection.

AlbertWeterings, 2018-02-21:
Went back to Arduino Core 2.3.0. this solves the problem.

closed 2018-02-21 by AlbertWeterings