xoseperez/espurna #568: How to implement a Mixed PWM and SPI Light provider

TomHodson, 2018-02-14:
Hello, first of all, I love this project thank you for putting it up and taking the time the help others use it on their devices. Scroll down to the bottom for a succinct question.

I currently have a custom build of Espurna flashed onto the board described in this thread: https://community.home-assistant.io/t/cheap-uk-wifi-bulbs-with-tasmota-teardown-help-tywe3s/40508/5

It has a TYWE3S module connected through SPI (CLK: IO4 DATA: IO14) to a SM1672 Led driver controlling the RGB channels and it also has white LEDs connected up to a second, single channel, LED driver that can be controlled by just PWM'ing pin 5. I can currently control the white channel via a PWM command that I added, but haven't gotten the RGB channels working yet because I haven't been able to find an english language datasheet for the SM1672. I suspect the protocol may be compatible with a more well known driver IC so I'm going to try using the FASTLED library next time I get a chance to try it.

Assuming that I do get it working, I would like to contribute back to Espurna by adding support for this board and sending you a pull request so my question is:

How would you like me to modify the code to incorporate a device with 3 channels that are driven by SPI and a 4th channel that is driven through PWM?

xoseperez, 2018-02-14:
Hi

Good work. I hope you can make it work. If you want to contribute to the project this is how I would do it:

Right now all the code for the lights is in the light.ino module. You should define a new LIGHT_PROVIDER in general.h and add the differing code in the light.ino file. Check how other LIGHT_PROVIDER_* are implemented (basically in: definitions, _lightProviderUpdate and lightSetup).

ericj2109, 2018-03-16:
@TomHodson Just wanted to see how this implementation was coming. I got one of these pesky bulbs a few weeks ago and followed along with that home-assistant.io thread. Dieter2000 there got it talking with FastLED and the SM16726 protocol so it's that much closer. I have been itching to flash the bulb and just wondered if you took a stab at it. Thanks!

TomHodson, 2018-03-16:
@ericj2109  I got most of the way there but never quite got the white pwm channel working, I just uploaded my modifications to my fork of espurna at https://github.com/TomHodson/espurna if you want to take a look!

If I get time to finish it I'll let you know.

xoseperez, 2018-03-19:
I have not read the datasheet for the driver. The only weird thing is that the white channel is driven differently (you are changing only the white channel PWM cycle). Is it supposed to be like that?

https://github.com/TomHodson/espurna/blob/201338ec8154250bec5217ff57a6201088cf7566/code/espurna/light.ino#L461

TomHodson, 2018-03-19:
Yeah, the RGB channels are driven by one driver, my code seems to work for that, although the bulb probably needs to heavy colour correction. 

The white channel is driven by PWM, which I confirmed with a call to analogWrite in an arduino sketch. I've tried to incorporate this into espurna but I haven't gotten it working yet, if you see any obvious mistakes in the way I've written the code your help would be much appreciated but otherwise I'm sure I'll figure it out eventually. 

There's a lot more discussion about this bulb at https://community.home-assistant.io/t/cheap-uk-wifi-bulbs-with-tasmota-teardown-help-tywe3s

xoseperez, 2018-03-19:
The pwm_set_duty uses _PWM which already does the mapping. You'd probably want to set the `LIGHT_MAX_PWM` to 1000 in general.h. for this provider.

ericj2109, 2018-04-09:
Finally got around to flashing my bulb! I'm not versed in the entire workings of ESPurna so my fiddling was pretty basic. Tried your fork @TomHodson and dimming was a little crazy, especially the white channel which should work pretty simply.  By removing the manual conversion of multiplying by 39 from light.ino and setting LIGHT_MAX_PWM to 255, I think the white dimming works properly but the color channels are hit or miss. Maybe I undid a correction you did for the RGB channels; I need to read up some more on that controller. Now I can only get colors to show with brightness set to about 130 and, even then, I need to send a color change twice to get it to take. Lower than 130 and (I believe) the right color and intensity flashes momentarily. Never actually tried the bulb with the stock firmware so, maybe my bulb is at fault. Interested to know what you find!

TomHodson, 2018-04-09:
I see the same issue with the color channels when I get rid of the * 39 and set LIGHT_MAX_PWM to 255.  Do the colors look ok when you flash my fork unmodified? 

I'm pretty sure this is just some little thing to do with getting the scale for the PWM to be 0 - 10000 and the scale for the colors to be  0 - 255

phototobi, 2018-07-09:
Hi Tom, 
i was really happy i found you´ve forked espurna to get it work on the cheap tywe3s bulb i  got when i ordered a sonoff b1 bulb. I was really suprised i got it flashed on the tywe3s  and it seems to work as i can configure my broker.. etc. as i really love with my magic home controller and the sonoff stuff. 
Did you get it work with pwm and rgb? i only like to use it for little led info when somethings finished like  laundry ready. I was missing the relay/x/set topic to switch it on or any other mechansim. and 3 channels to set rgb. It would be really nice if you could help me. If mqtt is not working, telnet commands would be also really great.
So i´m  a newbie in iot arduino and that stuff but i really like to learn as much as i can

Best regards,

Tobias

TomHodson, 2018-07-11:
Sorry for the slow reply, I'm travelling at the moment! 

It's been a while since I last looked at this but as far as I can remember, it was possible to change the RGB values through the web interface with my fork of the code, although as @ericj2109  pointed out there were still some issues with the colour. Should be fine for just making an indicator light though. Does your bulb work through the web interface if you use my fork? Did you double check that you're using the correct espurna endpoints? My fork is really a tiny modification to espurna so the docs should still help.

If you end up diving into the code, it's worth looking at the diff to see what I had to change to get it working: https://github.com/xoseperez/espurna/compare/master...TomHodson:master

The fork is also pretty far behind right now, so it could be that some of the features you're expecting were introduced into espurna after I made the fork.

phototobi, 2018-07-11:
Hi Tom,
thanks for your answer.
I´ve tried to build the version of espurna you´ve forked. But the old dependecies from bitbucket are no more available so i´ve put your light.ino and harware.h to the actual version to build it. So after succsessfull upload i can reach the website but theres no slider or something to switch it on or change color. As i wrote i don´t need perfect color corretion just set a color RGB and switch it on. Maybe you have still the old binaries? What does endpoint mean? I choose the tywe3s " Environment TYWE3S_SM16726_RGBW_Bulb            [SUCCESS]" Anything i can check? Best regards

Tobias

TomHodson, 2018-07-12:
I can’t really remember but there may be more changes you need to make to
get the web interface. If you have a look at the link in my last message
you’ll see the lines by line changes that I made in each file. Maybe there
are some more files you need to modify?

You could also try using git to automatically pull in the 500+ new commits
but I think you would have to resolve some conflicts.

I’ll have a look around to see if I have the binaries still.

TomHodson, 2019-01-20:
> Hi folks, I also have a few of these bulbs I'd really like to get running.
>
> I finally got one open and confirmed that the white channel is driven by
> PWM and the color is driven by a SM726EB
> <http://chinaasic.com/product.jsp#item=series%23style=21%23id=104>
> protocol equivalent to the SM16716/SM16726. Using the data sheet and
> following the traces I found the following GPIO mapping:
>
>    - GPIO04 color SPI clock
>    - GPIO05 white PWM
>    - GPIO13 color enable (PWM?)
>    - GPIO14 color SPI data
>
> I spent some time trying to set up the build environment but couldn't get
> an ESPurna build under 500kb. I feel as though I probably have the wrong
> versions of some libraries or something. I'll put some more time into it
> trying to see if I can get @TomHodson <https://github.com/TomHodson> 's
> build working. I think the color issue @ericj2109
> <https://github.com/ericj2109> was talking about might be that GPIO13
> should be enabled when in color mode.
>
> In the vendor firmware you can't have both white and color active at the
> same time, I'm guessing this is due to power limitations. It does have a
> nice fade effect when changing between white and color so I'm guessing
> GPIO05 and GPIO13 are PWM and crossfaded when switching modes.

It sounds like you've already made more progress than I did, I hope you get
it working! I have to admit I abandoned the project after some frustration.

Rather than trying to use my build directly, I suggest you just take a look
at the diff of my build against the espurna as it was when I made the fork.
And use that to see roughly where you need to modify espurna to get things
working. Be warned though, my version never really worked very well.

One thing that helped was to compile a very simple arduino sketch for the
ESP that bitbangs the relevant protocols on the relevant GPIOs, this way
you can test your assumptions about the hardware before moving on to trying
to integrate it into espurna.