xoseperez/espurna #569: WEB UI: how about adding Admin / Settings / Factory reset ?

PieBru, 2018-02-14:
Hi, every time I want a guaranteed clean start I execute ``` esptool --port /dev/ttyUSB0 erase_flash ```, but most of the time a factory reset is enough. The web ui is quicker than telnet, and more handy to not-CLI guys/gals.
Just my 2 cents.
Piero

hyteoo, 2018-02-14:
I already added a pull request for this exact topic :)
[https://github.com/xoseperez/espurna/pull/558](https://github.com/xoseperez/espurna/pull/558)

K1nslayer, 2018-02-15:
+1  
I've run into the need for this when flashing new Sonoff B1 Lights.  They were unstable for some reason (likely me not erasing the flash first) Having to terminal in was a bit of a pain since Apple removed basic telnet from the default utilities.

xoseperez, 2018-02-26:
Pull request https://github.com/xoseperez/espurna/pull/558 has been merged into dev branch.

xoseperez, 2018-02-26:
This is staged for release.

closed 2018-02-26 by xoseperez