xoseperez/espurna #570: Web UI doesn't always follow actual relay/switch state

K1nslayer, 2018-02-15:
Hello.  First off let me say that Espurna is awesome, and that it has great potential as a platform for ESP8266 projects.  Thank you for your work, I've enjoyed following the development.

I've noticed that the Web UI doesn't always correctly reflect the relay/switch state (on the status page) when changed from an external source.

When using the scheduler it does not get updated.  However, when using Alexa to turn it on/off it does get updated.  So it looks like a bug.

xoseperez, 2018-03-02:
When that happens, does the relay toggle? Can you reproduce it and copy here the debug log?

K1nslayer, 2018-03-03:
Apologies, with the latest code I wasn't able to reproduce this.  Perhaps its fixed, or perhaps the repro is more complex.  I'll close this ticket and reopen it in the future if I see the behavior again.

closed 2018-03-03 by K1nslayer