xoseperez/espurna #572: Android IDE compiled ESPurna Sonoff Dual Instability

gd-99, 2018-02-16:
Hopefully this hasn't been covered elsewhere. I did look through some of the issues but didn't find a reference to my symptom. 

**My environment**
I set up a fresh portable Arduino IDE (v1.8.5) and the Arduino Core for ESP8266 is 2.4.0. Followed the wiki guide for [compiling using the Arduino IDE](https://github.com/xoseperez/espurna/wiki/ArduinoIDE). I compiled the current dev (compiles to version 1.12.4a) targeted at the Sonoff Dual R2.

**My Symptoms**
Everything compiles without errors, however once flashed to the Sonoff Dual, it is very unstable particularly after after the first 30 seconds getting into a crash and reboot cycle and then finally dropping into safe mode.

**My solution**
I found that in my fresh install, of the Arduino Core for ESP8266 v2.4.0 the lwIP board setting is set to v2 Prebuilt by default. By reverting this setting to v1.4 Prebuilt, my instability went away.

This is a screen shot of my Arduino board setting that worked for me:
[image]

Many thanks for this great bit of firmware.

gn0st1c, 2018-02-16:
sounds like "WDT reset when enabling OTA (only with lwIP v2)", i thought this was fixed.
https://github.com/esp8266/Arduino/issues/4028

gd-99, 2018-02-17:
@gn0st1c Yes reading issue #4028 suggested to me it was resolved. I haven't tried recompiling with lwIP set to v2 to see if I reintroduce the instability, however I have done a fresh erase and recompile from scratch setting lwIP to v1.4 and it again recompiled with no errors and the Sonoff is rock solid. No crashing.

I will close this observed issue as I have a solution that works for me.

xoseperez, 2018-02-26:
Reopening, this should be properly documented.

gd-99, 2018-02-27:
@xoseperez I have also found that intermittently I lose connection to the Sonoff Dual R2 after an indeterminate amount of time, using either the web or telnet interface. My symptom appears similar to the issue here [Esp8266 IP Address not reachable after a while #2330](https://github.com/esp8266/Arduino/issues/2330).

I can confirm that even if I can't access the Sonoff via telnet or web interface, ESPurna appears to be otherwise working correctly - relays that should be on /off are in the correct position and scheduling continues to work correctly. When I do get back in via the web /telnet interface, the device uptime is correct.

I am sorry to be so vague, as it is intermittent I have not been able to pin the issue down with any certainty. I do get the feel it is related to lwIP and with the re-opening of this issue #572, think it reasonable to note this observation. Happy to raise this as a separate issue and link it to #572 if you would prefer / think there is merit in this recent observation? 

Workaround:
As a non perfect workaround that generally works for me, is to leave Telnet enabled. However after some time even with Telnet enabled I still lose connectivity. I have found that after repeated telnet connection attempts, telnet will respond with a connection, after which I am then also able to access the web interface.  I should note I am only accessing the device while on the local 192.168.x.x network.

gn0st1c, 2018-02-27:
i wonder if it's reporting WL_NO_SSID_AVAIL.

@xoseperez : how about a dump option on, say, a long key press + a send dump file option to webgui? 

thus, if connection is lost to the device but it otherwise keeps functioning, we can dump to SPIFFS.

xoseperez, 2018-03-02:
ESPurna does not use SPIFFS and enabling support for it adds up quite some Kb to the binary.

xoseperez, 2018-03-04:
I have added a comment and updated the suggested configuration for Arduino IDE in the wiki page.

closed 2018-03-04 by xoseperez