xoseperez/espurna #573: Scheduler enhancement: Set relay(s) to last active schedule after reboot

gd-99, 2018-02-16:
As I understand the scheduling function on my Sonoff Dual R2, the scheduler only "looks" ahead. That is to say, the firmware tracks up and coming active events based on the current time, setting the relays when the schedule is triggered.

For my usage it would be very useful, after the device reboots, after say a power cut, for the scheduler, on start up, to "look" back in time and reset the relay(s) to the most recent last active schedule. Once the retrospective check completes to go back to the current scheduling routine.

Happy to explain further if this is a do-able or useful enhancement.

Kind regards.

xoseperez, 2018-02-16:
ESPurna already has the option to set the relays to the last known state after a reboot. If a scheduled event changes the relay state, this new state will be stored in EEPROM and retrieved back after boot. I think this is a more generic approach.

Tardymo, 2018-02-16:
Will not work if state should have changed, but the device was off and is swithed on afterwards.

gd-99, 2018-02-17:
@xoseperez  Many thanks for the pointer. I don't know why I didn't think of this as a solution to my scenario. I have conducted some tests and it works perfectly for me - and as you point out a much better generic approach.

closed 2018-02-17 by gd-99