xoseperez/espurna #576: Hostname Maximum Character Length 14?

jlwainwright, 2018-02-17:
I have noticed that the maximum character length for the hostname has decreased to 14 characters somewhere between 1.10.0 and  1.12.1. I do not see a mention of this on the changelog. 

Was this an intentional change? Would be good if I can keep my previous naming convention that included my full room names and IP addresses.

Thanks for a awesome piece of firmware!

xoseperez, 2018-02-26:
There was no limit prior to 1.12.1 and now it has been fixed to 14 chars. That is the width of the web UI menu for 100% font size. I know it's all about aesthetics. Allowing it to be larger while keeping the name within the menu width is possible adding spaces or hyphens to the name.

xoseperez, 2018-03-12:
Staged for release

closed 2018-03-12 by xoseperez