xoseperez/espurna #577: RF Bridge enhancement: Generate RF codes for learning remotes

mr-sneezy, 2018-02-18:
Please disregard if this function is already planned, and also perhaps this need is so rare that it's not worth consideration anyway :)
I bought a few universal style 433Mhz 4-button remotes to use with the RF Bridge, but the remotes arrive with no user codes programmed (as I would expect, because they are universal replacements). As I had no old remotes to 'learn' from I had a problem now.
I worked around the problem by creating and typing a custom RF code in the Espurna RFBRIDGE web GUI page and saved them, then I put the 4 button remote into 'learn' mode and sent it the custom code via RF Bridges STATUS page button. This worked fine, so now the learning remote has a code and RF Bridge can obviously receive it back.
The espurna enhancement would be to auto generate a new RF code for a switch in the RFBRIDGE GUI that can then be 'learned' into a remote. Maybe a GUI button called GENERATE to the RHS of FORGET in the web GUI.

xoseperez, 2018-02-18:
I think this is more or less the same as #229

mr-sneezy, 2018-02-19:
Agreed, it's effectively the same feature request but perhaps #229 is with more tunability.

closed 2018-02-20 by xoseperez