xoseperez/espurna #579: Sonoff RF-Bridge problems

baddle, 2018-02-18:
Hello,

I'm trying to flash the latest espurna version to a Sonoff RF-Bridge and have problems connecting to the wifi.

1. turn bridge to off
2. connect bridge to programmer while holding the button
3. erase flash with esptool
4. flash espurna 1.12.3
5. disconnect after successful flashing
6. turn bridge back to on
7. connect to micro usb power
8. waiting for ESPURNA_XXXX AP and connect to it
9. change admin/wifi-password
10. reconnect with new credentials
11. add my wifi-ap, save and reboot

The bridge now tries to connect and the wifi light blinks for ~35s and I get mutiple times the following serial-output.

```
>>>stack>>>
3ffffd10:  40201864 00000000 3fff3914 00000000
3ffffd20:  40104be0 004d667c 3fff2160 00000000
3ffffd30:  3ffef4c0 3fff2160 3fff3914 4024a4c4
3ffffd40:  00000000 001db774 402151bc 00002710
3ffffd50:  3fff2160 00000000 00002710 00002710
3ffffd60:  00000000 3fff0df1 3ffffdc8 40201ad5
3ffffd70:  3fff38b4 00000000 00000bef 00000030
3ffffd80:  00000001 00000023 7fffffff 00000002
3ffffd90:  00000123 3ffee770 3fff2c24 00000001

3ffffda0:  3fff3914 0000007b 3fff3944 40201b08
3ffffdb0:  00000000 4023b5d3 00002200 40202df1
3ffffdc0:  0000007b 3fff3944 3ffea980 00000000
3ffffdd0:  3fff3914 3ffffe30 3ffffe30 40213c53
3ffffde0:  3fff392c 00000030 3fff3944 4020e034
3ffffdf0:  ec0600e3 00000000 00000000 34314e31
3ffffe00:  00000000 00000000 00000000 00000000
3ffffe10:  00000000 00000000 00000000 00000000
3ffffe20:  00000001 000005db 3fff0e64 4020e409
3ffffe30:  3fff3914 0000000f 0000000c 00000001
3ffffe40:  00000010 3ffffea0 3ffffea0 40213c53
3ffffe50:  3fff389c 0000000c 3ffffea0 3fff0e64
3ffffe60:  3ffea980 00000000 000001bc 00000001
3ffffe70:  00000001 00000000 3fff0e64 00000001
3ffffe80:  00000001 3fff3944 3fff0e64 4020e560
3ffffe90:  00000001 00000000 3fff0e64 4020e5de
3ffffea0:  00000000 00000000 00000000 402266a0
3ffffeb0:  00000001 00000000 3ffffee0 40223f48
3ffffec0:  3ffe8d40 44454c5b 754e205d 3ffef658
3ffffed0:  3fff0df4 3fff0e64 00000000 4021bf9e
3ffffee0:  3fff38fc 0000000f 00000001 3fff38e4
3ffffef0:  0000000f 00000006 3fff38cc 0000000f
3fffff00:  00000001 3fff38b4 0000000f 00000009
3fffff10:  3fff389c 0000000f 0000000c 3fff6dbc
3fffff20:  0000000f 0000000a 00000000 4021062a
3fffff30:  3fff0df4 3fff70a4 3fff3644 4021c090
3fffff40:  3ffef4c0 00000000 3fff3644 402018cc
3fffff50:  3ffea980 a105a8c0 3ffea980 00ffffff
3fffff60:  3ffea980 0105a8c0 00000000 00000005
3fffff70:  00000002 3fff3644 3fff70c4 40201b63
3fffff80:  401007d2 00000002 00000000 3ffef658
3fffff90:  3fffdc10 00000000 3fff3644 40245620
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<
FX
�␘��
Exception (9):
epc1=0x401049b4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000003 depc=0x00000000

ctx: sys
sp: 3ffffb70 end: 3fffffb0 offset: 01a0
```
After ~35s the wifi led stops blinking and change to a constant light blue and a ESPURNA_XXXX AP  appears again.

I can connect to the ESPURNA_XXXX and the AP assignes me the IP 192.168.4.2 but the bridge (192.168.4.1) refuse any connection to access the webinterface.

I tried various combinations and flashed dozens of times but the result is every time the same. Maybe the bridge-board is to new and unsupported (added pictures of the board)? I was able to flash Tasmota without a problem.

Thanks for help in advance.

[image]
[image]

franki29, 2018-02-18:
Hi,
i have the same on and i had no problems. Did you try to open the web interface with user name admin & passwort fibonacci ?

franki29, 2018-02-18:
Strange, maybe you try a older Firmware?

baddle, 2018-02-18:
@franki29 

I tried with latest 1.12, 11.11 and 1.10 with the exact same result.

```
Exception (9):
epc1=0x401049b4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000003 depc=0x00000000

ctx: sys
sp: 3ffffb70 end: 3fffffb0 offset: 01a0

>>>stack>>>
...
<<<stack<<<
```

AlbertWeterings, 2018-02-18:
Did you set flash mode to dout

mr-sneezy, 2018-02-19:
This seems similar to issue #572, might be worth trying the same fix, it worked for my issue too.

baddle, 2018-02-19:
@AlbertWeterings 

I flashed with Platformio with default settings for RFBridge:
```
[env:itead-sonoff-4ch-pro]
platform = ${common.platform}
framework = arduino
board = esp01_1m
board_flash_mode = dout
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m} -DITEAD_SONOFF_4CH_PRO
monitor_baud = 115200
```
The flash-mode was set to dout.

@mr-sneezy 
Thanks for the link, I will try it this evening.

beho1der, 2018-02-21:
Did you try reading RF codes?

armandogmendoza, 2018-05-10:
it turns out that it connects to the wifi if you provide a fixed IP e.g. 192.168.1.88 and the gateway e.g. 192.168.1.1, perhaps there is a DHCP issue?

xoseperez, 2018-05-13:
Just to recap, three notes:
1) Always use flash mode DOUT, as @AlbertWeterings says
2) Try using lwIP 1.4 instead of default one when building against Core 2.4.X
3) The board in the pictures is R2, I have not test it, maybe there is some hardware issue?

wildwiz, 2018-06-02:
I have a couple of R2 myself, and they work fine with espurna 1.12.6
here are the commands I use with esptool, which, by the way, are the same I use on R1 boards.
```
esptool.py --port /dev/ttyUSB0 write_flash --flash_size 1MB --flash_mode dout 0x00000 .pioenvs/itead-sonoff-rfbridge(-direct)-ota/firmware.bin
```
and there the platformio line to build and deploy via ota for updates
```
ESPURNA_AUTH=<my password> pio run -e itead-sonoff-rfbridge(-direct)-ota --upload-port=10.0.0.170 -t upload | grep -v SKIP
```

xoseperez, 2018-06-04:
@baddle Any update on this?

closed 2018-12-06 by stale[bot]