xoseperez/espurna #580: Sonoff RFBridge does not send code

franki29, 2018-02-18:
Hi,
i have flashed the new HEX file for the Bridge from https://github.com/Portisch/RF-Bridge-EFM8BB1 and try to send code.
Receiving and learning works with the new firmware for the EFM8BB1, but it does not send a code out.
I am trying this code 2724014003D4051511 but nothings comes out (Tested with pilight on a raspberry pi and of course with my power plug). Is there any limitation in the source code that prevent the bridge to send the data?

xoseperez, 2018-02-20:
I can't tell.
The integration with @Portisch firmware was done by @rhx based on his fork of Portisch work.

AlbertWeterings, 2018-02-22:
I have been testing this issue ESPURNA is not adding:
RF_CODE_START
RF_CODE_RFOUT_BUCKET
In front of the raw message and does not close the message with
RF_CODE_STOP

If you add AAB0 in front off the raw message and 55 at the end it will send the raw code.

You can also replace the code in rfbridge.ino line 135 until 149 with this code and the bug is solved.
`
void _rfbSendRaw(const byte *message, const unsigned char n = RF_MESSAGE_SIZE) {
    Serial.write(RF_CODE_START);
    Serial.write(RF_CODE_RFOUT_BUCKET);
    for (unsigned char j=0; j<n; j++) {
        Serial.write(message[j]);
    }
    Serial.write(RF_CODE_STOP);
}

void _rfbSend(byte * message) {
    Serial.println();
    Serial.write(RF_CODE_START);
    Serial.write(RF_CODE_RFOUT);
    const unsigned char n = RF_MESSAGE_SIZE;
    for (unsigned char j=0; j<n; j++) {
        Serial.write(message[j]);
    }
    Serial.write(RF_CODE_STOP);
    Serial.flush();
    Serial.println();
}
`

franki29, 2018-02-23:
Hello,
thanks for help, but still no success. If i add AAB0 and 55 at the end, i just get a beep at my bridge.
I have two of them, both with the same problem.
Thanks and best regards

AlbertWeterings, 2018-02-23:
@franki29 do you see the RF led blink? And do you send your code over MQTT to the bridge or from the web interface? I tested sending via MQTT based on your info I will try to reproduce your issue

AlbertWeterings, 2018-02-25:
Well I gave up, on lt for now. The firmware of @rhx and @Portisch is going to need some work. Not only sending RF codes doen't go very well at the moment. I have flashed multiple versions found at https://github.com/rhx/RF-Bridge-EFM8BB1 and https://github.com/Portisch/RF-Bridge-EFM8BB1 With both I was still able to receive correct codes from a EV1527 remote but receiving codes from a PT2264 remote completely stopped working.

A other downside effect is that somehow out of nothing it seemed to receive over 400 codes in less than 15 min where it only received about 100 codes in the past two months.

gn0st1c, 2018-02-25:
thank you for the information @AlbertWeterings
i'm too, waiting for efm8bb1 firmwares to be mature. until then i'm happy with [RFLink.](http://www.rflink.nl/blog2/)

Portisch, 2018-03-09:
I am currently working on this problem:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/2

franki29, 2018-03-18:
Using newest code from Portisch (Great work) with my older RFBridge solve the problem for me.
Using the newer bridge layout (R2 V.1.0), the same code from Portsich does not work. If anyone has the information what was change ( I saw a new U11 on the PCB) please tell us.

franki29, 2018-03-22:
So , the new chip seams to be a 12V "regulator" and the sending ship has change to a F115. At my bridge it looks like the F115 is broken. I solder it out and connect a cheap 433MHZ sender and now my new bridge is working too.

closed 2018-03-22 by franki29