xoseperez/espurna #581: WPS wifi configuration

serra82, 2018-02-19:
Is possible configure the WiFi by WPS?

xoseperez, 2018-02-20:
Not yet. But hopefully soon.

xoseperez, 2018-05-29:
Working version using v2 of the JustWifi library. But WPS increases the binary image more than ~20kb!!
Check the justwifi2 branch and use `-DJUSTWIFI_ENABLE_WPS`. Testing it right now on several devices and running fine.

xoseperez, 2018-06-11:
Merged to dev branch. It is NOT enabled by default. WPS increases the binary size a lot. Besides, from the security point of view it's not a good idea (disable your router WPS feature if you can). Anyway you can enable it passing the `-DJUSTWIFI_ENABLE_WPS` build flag. Staged for release.

closed 2018-06-11 by xoseperez

serra82, 2018-07-12:
I'm testing wps, added -DJUSTWIFI_ENABLE_WPS and:
#define BUTTON1_LNGCLICK    BUTTON_MODE_WPS
the led start to blink fast, I push the wps button on router, but the smartplug don't connect to the router.

Can you explain the steps to use wps?
Thank you