xoseperez/espurna #583: Batch OTA Upload, backup and more

soif, 2018-02-20:
I've just written [EspBuddy](https://github.com/soif/EspBuddy) , a wrapper script that ease the following tasks  (for a single Espurna, or in a batch)
- Upload via OTA
- Upload via Serial (with optionnal prior flash_erase)
- Compilations with usefull -D flags and pattern substitution
- Remote Espurna version query
- Archiving of latest + previous firmwares per devices
- Archiving of latest + previous settings per devices
- Parse local repo version
- Git Pull local repo version
- Ping
.....

Comments and PR are welcome at:
https://github.com/soif/EspBuddy

Enjoy 👍

xoseperez, 2018-02-20:
Great, thank you. Will add it to the wiki.
There is also an OTA manager in ESPurna, albeit not very documented. You can check the ota.py script in the root folder.

soif, 2018-02-21:
you're welcome
HTH

soif, 2022-02-15:
Hi since this old post, I've released a few versions (latest is v2.20) **with tons of enhancement**.

The current [v2.21 develop version](https://github.com/soif/EspBuddy/commits/develop) even handles sending telnet commands to Espura., ie:

`espbuddy.php send 192.168.1.1 uptime  --repo=espurna` returns the result of the _uptime_ telnet command...

Enjoy 😄 

_BTW if you still like it, feel free to add it to the Wiki, to let others Espurna users to also enjoy it._