xoseperez/espurna #584: factory reset vs several on/off cycles

kuchod, 2018-02-20:
After an electrical problem I had to reconfigure half of my sonof basic, because they reset to factory defaults if you cycle the power several times. Is this the expected behavior?

xoseperez, 2018-02-20:
Of course not. Are you sure the EEPROM was erased?

kuchod, 2018-02-21:
I have 6 sonof installed around de house and several of them where on (relay off) during the problem. 

The problem was some sort of derivation to earth in the kitchen that caused the differential breaker to open the circuit. Tried to rearm it several times until I realized what the problem was, and then the connected sonof's apeared with factory settings, every configuration erased. The disconnected ones where ok.

During reconfiguration, I think I entered a wrong password two times in one of the sonof's and lost access, so I went to the interruptor and cycled it several times (9?, 10?)... et voila! factory settings again. I dont know how many cycles does it need to reset, but until now it didnt happen with a normal use.

xoseperez, 2018-02-26:
Tried to reproduce it with no success... tried it cutting power for a short time or using longer times, tried it 9, 10, 15, 20 times and nothing. If the number of resets is 5 it enters safe mode (as expected), but aside from that, I couldn't make it go back to factory settings...

The default firmware does factory reset using the command line option (`factory.reset`) or holding the main button down for >10s (on most devices).

h3lmutek, 2018-05-01:
it does to me too :(

Bochets34, 2018-06-16:
Hello,

just got similar issues, my configuration is 12 sonoff dual R2, one plug S20 and one basic.
espurna release was 1.12.3.
only half (6 over 12) of my dual R2 got the issue: reset to default configuration: passwd, wifi and even httpi key changed.
I had to enter values again to get my setup back to 'normal'.

it appeared after an electricity cutoff for several hours last week. power came several time very briefly, trying to reengage from the provider, I'm leaving in countryside :)

I upgraded to the latest release 1.12.6, hoping it will prevent from that issue in the future...
Olivier

kakaki, 2018-08-03:
I have same issue with one device with nodemcu, it keeps resetting all configuration, after some days, couldn’t find what is causing problem, it is on 1.13.0c firmware.

kakaki, 2018-08-18:
Updated all devices to 1.13.1 - problem still exists - one of devices cleared all settings.
This time it's on esp8285 esp-m3 .

xoseperez, 2018-10-07:
have been abusing a Sonoff Basic hard. Cycle resetting it several (up to 20, maybe 30) times. I have done it several times, as fast as I could manually or waiting a few seconds in between. I couldn't make it fail. 

Anyone here is able to reproduce it in a consistent way?

rizkypr, 2018-12-22:
i have been using espurna on Sonoff Bbasic, this also happened to me too if i powered up the circuit breaker after a fault then if the circuit breaker is being trip immediately after power up again because of the fault all my sonoff basic are reset to default.

jlwainwright, 2019-01-17:
Same issue here. I have 20 Sonoff Basic's installed and this would occur when the trip switch gets switched on and immediately trips again. So basically a milliseconds on/off switching of the units. 

Checked and these devices were also on FW 1.12.3, 1.12.5 as with all my other devices.

mcspr, 2019-01-22:
@jlwainwright  @rizkypr bulk of this should be fixed in dev 
https://github.com/xoseperez/espurna/commit/89ed0cacebb359eccfbd70d17b07d52f96d060c1 for possible exc28/29 stack trace being too big
https://github.com/xoseperez/espurna/commit/7c85225764d98edce8596283d7fc72af35287811 doing less flash operations
also pending #1420 to avoid commits altogether for system stability checker and custom reset

eeprom commit() takes ~29ms with my d1mini (esp12e one) btw, tried testing with mechanical relay board, but idk if it switches that fast to test this.

closed 2019-02-21 by xoseperez