xoseperez/espurna #585: SENSOR CONFIGURATION on digital sensor over MQTT

eduardoalbarado, 2018-02-20:
Add individual / group "SENSOR CONFIGURATION" by sensor. For example, when adding a digital sensor, it changes state but dont send data to MQTT until the entire "SENSOR CONFIGURATION" interval is finished, which can be configured for 10 minutes to read temperature. Not letting have good data on opening, triggers, pulses. I hope to explain myself well. 
Regards  from Argentina

xoseperez, 2018-02-20:
Sorry @eduardoalbarado, I'm not sure I understand your request. Do you mean something like a sensor warm-up interval before start sending data?

gn0st1c, 2018-02-20:
this makes sense for MQ series sensors that require a preheat time.

eduardoalbarado, 2018-02-27:
Sorry for my English.
I'll try to explain myself better. If I need to monitor or take instantaneous actions based on data from a mqtt subcription of a digital sensor, these readings are dependent on the "report every" value in "sensor configuration". If I understand correctly, this setting affects all the sensors, and it gets complicated if each one requires different times.
It would be nice to have an immediate option, as is the case with relays over mqtt. I think it would be useful in fast action systems such as pressure valve systems or alarm systems.

xoseperez, 2018-02-27:
You mean sensors reporting based on events, instead of at fixed intervals? It sounds good.