xoseperez/espurna #587: ESP resets when load connected to HLW8012

varungujjar, 2018-02-20:
Hey @xoseperez  I created a board as per espurna schematics  .. everything works I also get exact current readings  however when a load is connected the device crashes and restarts .. it doesnt happen when no load is connected..  is there something am missing ..  ? tried with interrupt mode on and off for HLW8012 still no luck...  using a  Hi Link 3W AC DC convertor.. is that sufficient or does it need 5W..?

xoseperez, 2018-02-20:
I will check the decoupling capacitor for the ESP8266.

varungujjar, 2018-02-20:
cool.. let me know if you need me try ..?

varungujjar, 2018-02-24:
@xoseperez  fixed it the 100 nf capacitors have to be placed way closer than i thought.. :)

xoseperez, 2018-02-26:
Cool. Closing the issue :smile:

closed 2018-02-26 by xoseperez