xoseperez/espurna #588: MQTT Split topic data into many topics

theonedemon, 2018-02-20:
Some problem in #260 
device: sonoff pow
firmware: 1.12.3 ( binary from release page )
Settings: 
- Use JSON payload
- MQTT Client ID empty
- MQTT Retain on
- MQTT Keep Alive 10
1) 
```json
{
    "current": "0.000",
    "voltage": "219",
    "power": "0",
    "reactive": "0",
    "apparent": "0",
    "factor": "0",
    "energy": "59",
    "time": "2018-02-17 21:31:17",
    "mac": "XX:XX:XX:XX:XX:XX",
    "host": "boiler",
    "ip": "10.1.1.16",
    "id": 1004
}
````

2) 
```json
{
    "app": "ESPURNA",
    "version": "1.12.3",
    "host": "boiler",
    "ip": "10.1.1.16",
    "mac": "XX:XX:XX:XX:XX:XX",
    "rssi": "-52",
    "uptime": "174696",
    "datetime": "2018-02-19 22:00:59",
    "freeheap": "20136",
    "relay/0": "1",
    "time": "2018-02-19 22:00:59",
    "id": 1041
}
```

3) 

```json
{
    "relay/0": "0",
    "time": "2018-02-19 22:04:43",
    "mac": "XX:XX:XX:XX:XX:XX",
    "host": "boiler",
    "ip": "10.1.1.16",
    "id": 1049
}
```

4) 

```json
{
    "vcc": "3160",
    "time": "2018-02-19 22:00:59",
    "mac": "XX:XX:XX:XX:XX:XX",
    "host": "boiler",
    "ip": "10.1.1.16",
    "id": 1042
}
```

xoseperez, 2018-02-20:
Same explanation as in #260. JSON payloads are grouped by time, they all have a group of sensor/relay/heartbeat data plus some metadata.

theonedemon, 2018-02-20:
I use [image] and create button with option:
name: button
topic: boiler/data
json parse: $.relay/0
public topic: boiler/relay/0/set

and indicator always red ( off mode ): 
```json
{"app":"ESPURNA","version":"1.12.3","host":"boiler","ip":"10.1.1.16","mac":"XX:XX:XX:XX:XX","rssi":"-55","uptime":"88585","datetime":"2018-02-20 23:15:26","freeheap":"20152","relay/0":"1","time":"2018-02-20 23:15:26","id":4071}
```json
with id=4071 recieved message id=4072 at the same time:

```json
{"vcc":"3149","time":"2018-02-20 23:15:26","mac":"XX:XX:XX:XX:XX","host":"boiler","ip":"10.1.1.16","id":4072}
```
other indicators ( voltage, apparent ) works correctly.
Is it possible to make all metrics come in one json-message?

xoseperez, 2018-02-26:
This is controlled by the MQTT_USE_JSON_DELAY and MQTT_QUEUE_MAX_SIZE setting in general.h. Messages within 100ms are grouped up to a max of 10 messages. Changing the MQTT_QUEUE_MAX_SIZE to something like 20 will probably group them.

theonedemon, 2018-02-26:
Sorry, but is solution not work
i tested on witty-cloud and set MQTT_QUEUE_MAX_SIZE = 20
```json
{"analog":"355","time":"2018-02-26 21:20:05","mac":"XX:XX:XX:XX:XX:XX","host":"witty","ip":"10.1.1.17","id":3007}
```

```json
{"app":"ESPURNA","version":"1.12.3","host":"witty","ip":"10.1.1.17","mac":"XX:XX:XX:XX:XX:XX","rssi":"-70","uptime":"307","datetime":"2018-02-26 21:20:13","freeheap":"19760","relay/0":"1","color":"#FF0000","rgb":"#FF0000","hsv":"0,100,100","brightness":"255","channel/0":"255","channel/1":"0","channel/2":"0","time":"2018-02-26 21:20:13","id":3008}
```

xoseperez, 2018-02-26:
These are two different messages, for the analog sensor and the heartbeat. Unless they happen within a time window of 100ms they will be sent separately. Please mind that JSON output does not mean that there will only be one message with all data in it. It just means that close-in-time messages will be grouped in a single message.

xoseperez, 2018-02-26:
This is staged for release.

closed 2018-02-26 by xoseperez