xoseperez/espurna #594: RF_SUPPORT configuration

spiderjn, 2018-02-22:
Hi
I put last firmware on a Sonoff RF device, it's work perfectly with MQTT/Domotics, now want also configure RF communication with my RFxCom (for use outside my WiFi range).
How i can set RF identifier inside device ?
Best regards

spiderjn, 2018-02-26:
Hi,
I see [RF] Enabled in log debug but i can't find any information on protocol, i test all switch avalaible on domoticz without success.
So i propose to make evolution on web interface for display RF code and an other for send code when state change (this can help to learn code outside).
Best regards

xoseperez, 2018-02-26:
RF support is very old and it has not been tested maybe in the last 2 years... so yes... it might not work as expected (or at all). It was added to the first revisions of ESPurna after this project: http://tinkerman.cat/adding-rf-to-a-non-rf-itead-sonoff/

spiderjn, 2018-02-27:
i understand, i've read your initial project but it's not based on standard itead RF module.
yesterday i found  a return in first intruction from Rf loop ...

xoseperez, 2018-02-28:
LOL. I guess I disabled it the hard way...

xoseperez, 2018-02-28:
I have upgraded support for the RF module, using the RFBridge web interface to learn new codes. Can you give it a try? It's in the dev branch, just set RF_SUPPORT to 1 and configure the receiver GPIO.

xoseperez, 2018-03-01:
Do a `git pull` to get the latest changes, there is no `return` in `rfLoop` now.

spiderjn, 2018-03-01:
ok fine, after flash, now i've in debug 👍 
[068694] [RF] Code ON : 266325
[068694] [RF] Code OFF: 266321
but where configure GPIO receiver ?

xoseperez, 2018-03-01:
You should not be seeing those message, are you sure you are in dev branch?

```
git checkout dev
git pull
```

spiderjn, 2018-03-01:
i'm in dev branch, but it an error during flashing.
1 : i change just this line in 
[env:itead-sonoff-rf]
build_flags = ${common.build_flags_1m} -DITEAD_SONOFF_RF -DITEAD_SONOFF_RF -DRF_SUPPORT=1 -DDEBUG_TELNET_SUPPORT
2 : new flash with cable.
3 : now i've correct firmware inside device 👍 
Firmware version     1.12.4a
Firmware build        2018-03-01 12:39:05

and now ?

xoseperez, 2018-03-01:
Configure the GPIO under the RF section in `general.h` file.

spiderjn, 2018-03-01:
you mean, change value :
#define RF_PIN 14
but what number ?

xoseperez, 2018-03-01:
The GPIO you have the RF module connected to...

spiderjn, 2018-03-01:
Oups yes, but where i can find this information ? on Itead schematic ?
https://www.itead.cc/wiki/images/f/f8/Sonoff-Schmatic.pdf

xoseperez, 2018-03-02:
The RF module works when you have the RF module connected to your ESP8266 microcontroller. If you are using the Sonoff RF it should work out-of-the-box without any special features. If it doesn't then it means the Sonoff RF does not support your specific remote. To pair them you have to double click the Sonoff button (the red light should lit) and then click the button in your remote.

spiderjn, 2018-03-02:
you'r right. when i use sonof button i see led changing color, but nothing in log.
i use an Itead remote received with device, and i can change relay status, but log is confusing and RF Web Panel seem's not working.
How i can help to troubleshooting this enhancement ?

xoseperez, 2018-03-02:
RF_SUPPORT is meant for devices where the RF module is linked to the ESP8266. On the Sonoff RF there is a secondary microcontroller that manages the RF module, so it's no use. Disable it. 

The only way to "learn" new codes with the Sonoff RF is to double click the Sonoff button and press the remote button to link that remote button to the Sonoff relay.

xoseperez, 2018-03-05:
Closing, changes to the RF module are staged for release.

closed 2018-03-05 by xoseperez