xoseperez/espurna #596: ECH1560 power meter - no data (((

marsic-84, 2018-02-22:
ECH1560 power meter no data (ESPURNA 1.12.3)

I have everything connected properly.
I tried this https://github.com/HarringayMakerSpace/WiFiPowerMeter
and i works fine.

Can you make ESPURNA to work with ECH1560 power meter?? and add more data like power factor and frequency??

gn0st1c, 2018-02-22:
just to make sure;
you're using espurna-1.12.3-esp01-ech1560.bin and you've connected CLK to GPIO4 and MISO to GPIO5 right?

are you getting any data at all?

marsic-84, 2018-02-22:
I was using latest zip from github and arduino IDE 1.8.5,  ...and i'm using esp12e, ...and i've connected CLK to GPIO4 and MISO to GPIO5 
and like i said it works with this firmware https://github.com/HarringayMakerSpace/WiFiPowerMeter
but with ESPURNA i see only zeros(((  
what else should I check??

gn0st1c, 2018-02-22:
afaik Xose never had the ech1560 version of the powermeter.

i'll take out mine from the storage and give it a try this weekend, see if i can make it work. will let you know.

marsic-84, 2018-02-22:
ok, thx, will wait))

marsic-84, 2018-02-22:
just try firmware from espurna-1.12.3-esp01-ech1560.bin file.
still the same((((
[000515] [SENSOR]  -> current:0
[000515] [SENSOR]  -> voltage:0
[000517] [SENSOR]  -> apparent:0

gn0st1c, 2018-02-25:
i can confirm that it does not work.

in ECH1560Sensor.h, _sync is never called.

xoseperez, 2018-02-26:
Hi, I received 2 ECH1560-based power meters 2 or 3 weeks ago but haven't had the time to play with them. As soon as I have the time I will test the sensor and fix it.

gn0st1c, 2018-02-27:
tip: _sync is not called when _dosync flag is set. (there is no if (_dosync) check)

xoseperez, 2018-06-21:
So far I have failed miserably on fixing this. I'm going to wait a couple of days. Moving forward.

karmacoma92, 2018-07-04:
Hi!
I have one of those power meters with KR-PM01-A8 board which seems to be based on ECH1560. I get values but weird ones, have tried with and without Inverted setting but no luck. If you like me to do any testing which could help let me know 
Thanks a lot for your work!

DaNii78, 2018-08-12:
I have the same board and it doesn't work..

SiskoBen, 2020-08-08:
Hi, I got an ECH1560 power meter to work with some modifications (adding reading the MOSI pin to know when the power meter is requesting V, U, I from the ECH1560). It works now and i can see the values in ESPurna (and post them to ThingSpeak (after realizing the ESPurna does not like to talk HTTPS to ThingSpeak but only HTTP)). 
Anyway, is anyone is interested to get my modified files I would be happy to post them. Mostly it is based on Karl Hagström his initial investigations and work and my own investigations and updates in 2016. I have no idea how to get this into a PR on GITHUB

mcspr, 2020-08-09:
@SiskoBen afaik, github allows file drag-n-drop everywhere, you can attach the file here / https://gist.github.com / drop in into a new branch (see fork)

Do you use multiple ECH1560 class instances?
Any reason why we can't use SPI class to read this stuff?

SiskoBen, 2020-08-09:
[espurna.zip](https://github.com/xoseperez/espurna/files/5048267/espurna.zip)
@mcspr i wondered about whether it would be possible to use SPI to read this stuff as well but i stuck with the spying/bit-analysis (of the communication between the powermeter control board and the ECH1560 chip on the power board) that Karl Hagström started with and which is largely what is currently in the ECH1560 code for ESPURNA.
I have attached a zip file with just the modified files, I had to modify a few of them as I've added a pin for mosi spying (default on gpio 15), i tried to be as true to the design philosophy/style of ESPURNA code. I don't even know if these power meters with the ECH1560 are still available as i bough these about 4 years ago (via AliExpress). I will most likely re-purpose the current hardware (powermeter with Wemos D1 mini) to program with IOTAppStory's framework as ESPURNA does not seem to be capable to send notification e-mails (yet) when measurements fall below or rise above certain levels (i'm trying to make a simple device for my father so that he gets an e-mail notification when his washer is finished as the build in notification sound it not loud enough to be heard).

jon-daemon, 2021-05-15:
I have this power meter.
[image]
I don't see any markings on the IC but I'm guessing it's a ECH1560.
I connected my power meter with a ESP8266 using optocouplers but it wasn't working. It wasn't showing data even on the lcd. I think that the schematic [here](https://tinkerman.cat/post/connected-power-meter/) is for active high signal and it's not working for ECH1560.
I don't expose any components outside of the case so I decided to remove the optocouplers and I connected CLK to GPIO4  and SDO to GPIO5 through 6k8 resistors. Now the lcd is working and I see some data but they are not correct.
[image]
[image]
[image]
I flashed with espurna-1.14.1-esp01-ech1560.bin. Is there any progress for my power meter?

fragtion, 2024-11-16:
Bump. How is support looking for ECH1560 ?

BenSeventy9, 2024-12-15:
Has anyone found a datasheet for ECH1560 chip?