xoseperez/espurna #598: KMC 70011 Support

WayneManion, 2018-02-22:
I bought three KMC 70011 energy-monitoring smart switches from Amazon for less than $30. 

https://www.amazon.com/KMC-Monitoring-Required-Control-Compatible/dp/B07313TH7B/ref=sr_1_3?ie=UTF8&qid=1519340130&sr=8-3&keywords=kmc+smart+plug+3+pack

I chose these because one of the reviews said there was an ESP8266 inside and that it works with Tasmota. I prefer Espurna, so I did some research and found out the 70011 uses the same power monitoring circuit as the Sonoff Pow. I dug through the Tasmota source and found the pin configuration inside the 70011. I added the following bit of code to hardware.ino:

```
#elif defined(KMC_70011)

    // Info
    #define MANUFACTURER        "KMC"
    #define DEVICE              "70011"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          14
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    0

    // HLW8012
    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT     1
    #endif
    #define HLW8012_SEL_PIN     12
    #define HLW8012_CF1_PIN     5
    #define HLW8012_CF_PIN      4
```

The device appears to work, power monitoring and all. I'm not terribly familiar with using Git, so I put the code here instead.

WayneManion, 2018-02-23:
I should add that flashing these devices is pretty simple. There is some kind of ESP-12 module soldered to a larger PCB. Just connect 3.3V and GND to a power supply of some kind and Tx and Rx to a serial adapter and off you go. The button is connected to GPIO0, so pressing it when you turn the device on puts it into programming mode.

xoseperez, 2018-02-26:
Thank you! Added it to the dev branch. Will be released with 1.12.4 soon.

xoseperez, 2018-02-26:
This is staged for release.

closed 2018-02-26 by xoseperez

edsai, 2019-05-06:
I just bought 3 of these from Amazon as well. I backed them up, erase the flash, flashed the pre-built kmc firmware and nothing. I took a (1MB) backup and tried to flash it back and nothing. Tried to flash a second device and nothing when I plug it in or when it reboots after flashing connected to the ttl interface. Curiously, my plugs say KT-70011B-KMC V1.0 instead of the 1.2 that's in the wiki. Not sure where to start debugging. There's no serial output when running serial monitor in Arduino at any baud rate. I tried to flash both 1.13.5 and 1.13.5 and no joy. Do you recall what release of the board you received?

jomeg0, 2020-06-21:
> I bought three KMC 70011 energy-monitoring smart switches from Amazon for less than $30. [...]

Hi mr. Manion, I saw in the amazon comments/review that you managed to flash the Xenon Touch switches. I have one of this (a 2 gang) can you send a wiring which to buttons do I have to push in while connecting to USB to bring it in flash mode? Many thanks in advance. Any help would be appreciated