xoseperez/espurna #602: Espurna 1.12.3

gokalpakgul, 2018-02-24:
Dear All

With Espurna 1.12.3 firmware is it possible that relay switches to OFF mode if the MQTT connection is broken for the safety reason? I saw MQTT retain option but no MQTT OFF option. Is there a plan to implement such a function?

Thks for your answers in advance

Gokalp

xoseperez, 2018-02-26:
I have never thought about it. If the MQTT disconnects it should reconnect as soon as the WiFi connection is restored or the broker available again... what kind of safety reasons are you thinking about?

gokalpakgul, 2018-02-27:
Assume that espurna is ON by the broker and after some time connection is broken. Because of wireless connection this can happen if wireless signal is weak. So we have the issue that Espurna stays ON. So if there is a setting in web server for MQTT such that if the connection is broken e.g. for 5 minutes if Espurna should switch to OFF this issue can be solved.

mcspr, 2018-03-02:
@gokalpakgul did you consider unconventionally using pulse mode? Set it for some small delay and just periodically send 1 to `sonoff/relay/0/set` Pulse timer will restart every time it receives mqtt message and will turn off when it misses one.

torwag, 2018-05-06:
That the timer will reset is a good start. I was looking exactly for the same functionality. Some application require that the relay switch off as a safety feature... e.g. if the broker crashes or wifi is borked, a gardening watering pump, which is suppose to run for 30 min a day, would run endless possibly creating larger damage to garden and buildings. Thus, we need a safety feature. If there is no connection and no response from the server side for e.g. 5 min, the pump should switch off.

mcspr, 2018-05-07:
@torwag For device-only option there's also scheduling function that uses network time to turn on / off relay - see Scheduler on web interface.
i.e. you can schedule it for specific time of day and set pulse timer to 30 minutes. But there are also couple of issues with this approach:
- it will not trigger if the time is missed. for example, ntp clock borked and changed from 19:01 to 19:03 when schedule is set to 19:02 - nothing happens.
- pulse is bound to device runtime, thus rebooting the device when timer got 15 of 30 minutes left and restoring last relay status (espurna can remember relay state between reboots) does not restore the timer - it will run for 30 minutes again.

If network is fairly stable, short pulse time and periodic keep-alive using relay/set does the thing. You can have 90 second pulse and 60 seconds interval between sending /set - 29 times and let the pulse timer run out to turn the relay off.

torwag, 2018-05-07:
Thanks for the comments. I use the same method as @fpalinf ... pulse mode with boot mode off.
As for the timer... it really should not miss a scheduled time... looks to me like a == has to be replaced by a >= together with the introduction of a small simple state-machine.

I will further investigate and see if it works as I need it to work. Thanks again for this great project.

xoseperez, 2018-05-08:
See also #845.

closed 2018-06-04 by xoseperez