xoseperez/espurna #603: sonoff basic dht

Vadonat, 2018-02-25:
Hello!I'm new in here.(sorry for my english if is too bad..).I have two pieces dht22 sensor on two sonoff basic with the latest sonoff basic dht rom.It is working but the sensor randomly goes to timeout.Plug out the cable from the wall is solve the problem.worked a day and after did the same.plug out-plug in solved the problem again but repeating after randomly again.I have a resistor too between the vcc and the data.and the wires are solderet to board directly.pls help me.thank you.

xoseperez, 2018-02-26:
This is hard to debug. The DHT sensor is stateless, it doesn't have any startup configuration, so if it goes offline and doesn't come back it has to be something external to the sensor code (hardware, interrupts,...). Is it happening with both sensors?

Vadonat, 2018-02-27:
yes.two sensors reproduce it.sometimes working a day,a week,3-4 hours and randomly do this.the devices are Aosong am2302.

K1nslayer, 2018-02-27:
I have 2 DHT22's working on Sonoff Basic hardware.  I'm compiling my own firmware and using these:
[https://www.amazon.com/gp/product/B073F472JL/ref=oh_aui_search_detailpage?ie=UTF8&psc=1](url)

Vadonat, 2018-02-27:
Can't open the link.I'm not a programmer.I just uploaded this awesome rom.:)

K1nslayer, 2018-02-27:
Fixed Link.  I uploaded 1.12.3 from the release dir to one of my basics.  If it times out, I'll let you know. 
[DHT22 @ Amazon](https://www.amazon.com/gp/product/B073F472JL/ref=oh_aui_search_detailpage?ie=UTF8&psc=1)

Vadonat, 2018-02-27:
ok.thank you!

Vadonat, 2018-02-27:
mine is the cheaper version.

Vadonat, 2018-03-01:
I tried it on 5 volt.there is no timeout now.

xoseperez, 2018-03-01:
It's a bad quality sensor then, it should work at 3V.

Vadonat, 2018-03-02:
I will buy normal quality sensors of my next salery..this is a quality issue I think.

closed 2018-03-02 by xoseperez