xoseperez/espurna #604: Extend LED mode: Mixed ON mode

hyteoo, 2018-02-25:
I'm currently working on several device types including single relay switches inside various devices.
An example would be ESP-01 based relays found on ebay (ex. [http://ebay.to/2opYkTC](http://ebay.to/2opYkTC))
For simplicity's sake and also limited pins to deal with, I usually prefer having 1 indicator LED instead of 2 like sonoffs. I usually use a switch with build in LED.

I also want wifi status to be indicated by the LED.
For this, a new mode would be welcomed based on Mixed but reversed.... LED on when relay is on and viceversa.
Also when turning on/off, LED should instantly respond (currently in mixed mode, the LED toggles a few seconds after the relay is toggled)

Here's an example:
[image]

xoseperez, 2018-02-26:
Hi
I have added a new mode to your requirements and renamed some others trying to make them more understandable. "Relay & WiFi" mode will stay ON if any relay is ON and OFF otherwise and the WiFi status will blink on top of that. The "Find me & WiFi" will do the opposite.

hyteoo, 2018-02-26:
That is great (and fast :). Naming looks more straightforward now.
Can't wait to test that.

xoseperez, 2018-02-26:
You can already do it using the dev branch to build and flash your device.

xoseperez, 2018-02-26:
This is staged for release.

closed 2018-02-26 by xoseperez

hyteoo, 2018-02-28:
Tested this, works correctly
However there is a 3 second delay between the time the relay switches and the LED is toggled.
It doesn't respond like for example "Relay Status". Can you specify where is this delay defined?

xoseperez, 2018-03-05:
This is due to the way the WiFi notification works. It's something to change in the future. Basically, it will "recalculate" the blink mode after the current blink cycle which is 5 seconds...