xoseperez/espurna #607: Support for Tuya Smart Life App

varungujjar, 2018-02-26:
Hey @xoseperez  was surfing the internet and the stumbled upon smart life app and it looks amazing they also have sdks to support your own custom device... may be am asking for too much :P but is there a way to add the espurna board support to the smart life app.. ?

https://play.google.com/store/apps/details?id=com.tuya.smartlife&hl=en

xoseperez, 2018-02-26:
They look like having branded ESP8266 modules inside... Where can I find the SDK for these devices?

varungujjar, 2018-02-27:
You can logon to this website https://en.tuya.com/  and signup for free..  if you scroll down you'll see a Client Guide..  just incase i downloaded a sample sdk by filling in the information on their site .. have attached the same..
[mcu_sdk_ESP_Smart_Power_Switch_20180227.zip](https://github.com/xoseperez/espurna/files/1761553/mcu_sdk_ESP_Smart_Power_Switch_20180227.zip)

xoseperez, 2018-07-01:
Sorry, out of the project scope.

closed 2018-07-01 by xoseperez