xoseperez/espurna #608: MQTT Button event seems wrong

pcornill, 2018-02-26:
Hi Xose,
I have setup a button as follow and I was expecting an event suite like 1/2/1/2 when switching ON/OFF/ON/OFF but instead I always get event '2'
(I use release 1.12.3)

in hardware.h
#define BUTTON2_PIN         16
#define BUTTON2_MODE        BUTTON_SWITCH | BUTTON_DEFAULT_HIGH

from console:
[392453] [MQTT] Sending ESPURNA_17E8B5/button/1 => 2 (PID 51)
[392531] [MQTT] Publish ACK for PID 51
[395507] [BUTTON] Button #1 event 2
[395509] [MQTT] Sending ESPURNA_17E8B5/button/1 => 2 (PID 52)
[395575] [MQTT] Publish ACK for PID 52
[397421] [BUTTON] Button #1 event 2
[397423] [MQTT] Sending ESPURNA_17E8B5/button/1 => 2 (PID 53)
[397494] [MQTT] Publish ACK for PID 53

pcornill, 2018-02-26:
I see, thanks for the clarification.

xoseperez, 2018-02-26:
You are welcome

closed 2018-02-26 by xoseperez