xoseperez/espurna #610: HLW8012 define default calibrated values

varungujjar, 2018-02-27:
Hey @xoseperez  is it possible to define the default calibrated values.. internally ? so we dont have to do the same for other devices.

xoseperez, 2018-02-27:
ESPurna uses the `HLW8012_CURRENT_R`, `HLW8012_VOLTAGE_R_UP` and `HLW8012_VOLTAGE_R_DOWN` values to calibrate the sensor. They are defined in the `sensor.h` file unless they are defined before, in the `hardware.h` file, for instance. Check the `setResistors` and  `_calculateDefaultMultipliers` methods in the HLW8012 library.

gn0st1c, 2018-03-21:
@varungujjar : an example can be KMC_70011 (see hardware.h)

shouldn't this issue be closed already?

varungujjar, 2018-03-21:
@xoseperez  yes this should be closed..

closed 2018-03-21 by varungujjar