xoseperez/espurna #613: Espurna with push button

Mattleyu30, 2018-02-28:
hi xose
first congratulations to Espurna, an excellent firmware for sonoff, for the moment I have not could be tested products but I would like to start with the project that I would undergo.
I state that as a management software use Domoticz
to a Wemos D1 pro I would like to connect 3 push buttons on pins D5-GPIO14 / D6-GPIO12 / D7GPIO13, with these buttons I would like to activate relays' of a sonoff, but I would not switch from domoticz because the response times from the moment of pressure de button at the moment of closing the relay we vary from 1 to 3 seconds.
Is there the possibility of communicating directly with another Espurna without going from domoticz?
or is there a way to speed up communication?
I have seen that MQTT Group can be used to etify, but I can not find a guide that explains how to do it.

xoseperez, 2018-02-28:
MQTT groups is a (so to say) peer to peer communication between devices, albeit it needs a broker since the messages are MQTT messages. Basically, you set the same group topic on two or more devices and they will synchronize their states. But again, it is MQTT and sometimes it's instantaneous while other your can notice a slight delay, same as with Domoticz.

Mattleyu30, 2018-02-28:
Thanks for the quick answer,
ok all clear, just enter the same argument on both switches. however, however, there may be a delay due to the MQTT.
There are no other systems for faster communication?

xoseperez, 2018-03-01:
There are, out there, but not in ESPurna :smile:. Check #467

closed 2018-07-29 by stale[bot]