xoseperez/espurna #615: Sonoff TH - Problem temperature and humidity combined

Mattleyu30, 2018-03-01:
Hi, Xose
I loaded the firmvare espurna sonoff th 1.12.3 on sonoff th with dht22, so far everything ok but when I go to put the idx on temperature and idx on humidity (which are identical) the communication MQTT on domoticz arrives on 2 lines:
2018-03-01 17: 03: 18.552 MQTT: Topic: Domoticz / in, Message: {"idx": 11, "nvalue": 0, "svalue": "22.3"}
2018-03-01 17: 03: 18.658 MQTT: Topic: Domoticz / in, Message: {"idx": 11, "nvalue": 17, "svalue": "2"}
Even putting ON "Use JSON payload" does the same thing and therefore I can not see the correct data on domoticz.
I can not understand where I'm wrong

xoseperez, 2018-03-02:
This is not supported at the moment. Create different virtual sensors in Domoticz (one for temperature and one for humidity) so they have different IDX.

Mattleyu30, 2018-03-02:
ok Xose,
I was hoping that you could join the message in order to have only one virtual sensor in domoticz. For the moment I use it with 2 sensors.

Thanks

closed 2018-03-03 by Mattleyu30