xoseperez/espurna #616: Auto Discovery Home Assistant

RoSulek1, 2018-03-02:
Hi at all
Sonoff Basic in Pulse Switch Mode
Integration with HomeAssistant -> ON

All working fine with auto discovery mode, but i cannot find the configuration of this Sonoff in configuration.yaml
I would like to modify the configuration by myself, to create switch in configuration.yaml working in pulse mode.

When i can find this configuration in HA?

xoseperez, 2018-03-02:
The documentation in HA is misleading here. It says " The basic idea is that the device itself adds its configuration into your configuration.yaml automatically" (https://home-assistant.io/components/discovery/) but it does not. 

As a developer states [here](https://github.com/home-assistant/home-assistant/issues/9135) "The discovery component does not add discovered devices to your configuration.yaml file, it only makes them available in the UI without the need of any configuration. ", same issue at the bottom: "We are not going to have discovery write to the configuration.yaml file."

mcspr, 2018-03-02:
@RoSulek1 You can easily listen to the mqtt traffic via `mosquitto_sub`
```
$ mosquitto_sub -V mqttv311 -h <mqtt_host> -u <mqtt_user> -P <mqtt_pass> -t homeassistant/# -v
homeassistant/switch/ESPURNA_8E0067_0/config {"name":"ESPURNA_8E0067","platform":"mqtt","state_topic":"home/ESPURNA_8E0067/relay/0","command_topic":"home/ESPURNA_8E0067/relay/0/set","payload_on":"1","payload_off":"0","availability_topic":"home/ESPURNA_8E0067/status","payload_available":"1","payload_not_available":"0"}
```

json keys are directly mapped on [switch.mqtt](https://home-assistant.io/components/switch.mqtt/), so you could even write some script to grab this data and output yaml configuration. (`!include switch.yaml` inside switch domain, or directly just generate whole `configuration.yaml`)

xoseperez, 2018-03-02:
Commit 207f58f0e98eb00997f322baa23e9dc0de8776f3 adds a button to the web UI that displays the required settings for the device in Home Assistant. So now you will have two ways to register your device, automatically using MQTT Discovery feature or copy-pasting the settings right into your configuration.yml file.

xoseperez, 2018-03-03:
Staged for release

closed 2018-03-03 by xoseperez

RoSulek1, 2018-03-03:
Perfect!

U are my boss ;)