xoseperez/espurna #617: just a Question: emon - energy for plain analog input - DC bias filtering

gnkarn, 2018-03-02:
Jose , i would like to understand if   the analog interface for Emon uses the Emonsensor  ( sampling the sinusoidal signal)  , the reason I'm asking is because it seams it is not "filtering" the DC bias component of the input signal , and also by applying the calibration method ( using zero amps , with the CT out of the wire)  it doesn't correct either .

So in my case with Zero amps , it is always reading 14 amps - ( it is a 30 amps to 1 v sct013)

thank you

gnkarn, 2018-03-02:
i think i found the problem , will ask later when i reach more clarity, tks

xoseperez, 2018-03-02:
It is, check these lines:

https://github.com/xoseperez/espurna/blob/bf2ef37b60cac1a258b51309d435f575936cd16c/code/espurna/sensors/EmonSensor.h#L155-L172

gnkarn, 2018-03-03:
yes, thank you,

it is working , will try to make a calibration tomorrow .

xoseperez, 2018-07-01:
I guess it's safe to close it now.

closed 2018-07-01 by xoseperez